VC-MIPI-modules/vc_mipi_nvidia #4: Device tree not building?

ZalvinZ, 2023-03-21:
Hi I've tried to install the camera driver for IMX327 to my Jetson nano, but keep running into this error.

make[2]: * No rule to make target 'arch/arm64/boot/dts/..//hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a00.dts', needed by 'arch/arm64/boot/dts/ddot//hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a00.dtb'.  Stop.
make[2]: * Waiting for unfinished jobs....
  DTC     arch/arm64/boot/dts/ddot
cc1: fatal error: arch/arm64/boot/dts/..: No such file or directory
compilation terminated.
make[2]: * [/home/akbanshee/vc_mipi_nvidia/build/NanoSD_32.6.1/Linux_for_Tegra/source/public/kernel/kernel-4.9/arch/arm64/boot/dts/Makefile:133: arch/arm64/boot/dts/_ddot_] Error 1
make[1]: * [arch/arm64/Makefile:154: dtbs] Error 2
make: *** [Makefile:171: sub-make] Error 2

Any idea on how to solve this issue?

VC-Support, 2023-03-28:
Which Ubuntu version are you using on your host PC?

ZalvinZ, 2023-03-28:
Hi, I am using Ubuntu 18.04. Have tried with Ubuntu 20.04 as well.

andresmanelli, 2023-05-05:
Hey @ZalvinZ maybe this can help https://forums.developer.nvidia.com/t/failed-to-make-l4t-kernel-dts/116399/9