VC-MIPI-modules/vc_mipi_nvidia #6: Patch files fail to apply to 32.6.1

bfoz, 2023-05-08:


bazo80, 2023-07-12:
Hello bfoz,

the problem is carriage return/line feed related.
If you are using the gnu patch, you could convert the patch files with:
dos2unix file.patch
After that, the gnu patch binary should be succeeding

closed 2023-10-02 by pmliquify

ray-inb, 2024-03-21:
@bfoz did you apply this patch in Yocto build ?

andresmanelli, 2024-03-21:
@bfoz in 32.6 the nvidia Kernel modules live out of tree in the `nvidia-oot` directory.

So you need to apply them in the kernel or in nvidia-oot, as it fits.

Also, the csi stream write functions are not there in 32.6 so there is a patch that does not apply (you can add the missing code if necessary).

Beware that the nvcsi registers are protected behind a firewall in the default mb2 so you need to add some extra rules and reflash (you can find this in the forums)

Hope it helps.

I can share what I've done if anyone is interested, but I saw an open ticket to support 36.2 so maybe this will come soon.

ray-inb, 2024-03-21:
I am working on 35.4.1; yes, they have milestones for 35.4.1 and 36.x, but I do not know when they will be available. 

I am working with Jetson Orin Nano Devkit 8GB variation. I am building my custom image using Yocto-OpemEmbedded reference distribution. I have added the driver and device tree following  I[ntegrate the driver in your own BSP](https://github.com/VC-MIPI-modules/vc_mipi_nvidia?tab=readme-ov-file#integrate-the-driver-in-your-own-bsp) instructions. Later, I added my custom distro within the kernel recipe with the vc_mipi patches. It flashed without any error, but I could not see any traces of vc_mipi in the live image.

According to your last comment, I think I applied them in the kernel using the bb append file.  Any idea am I doing it right ?

bfoz, 2024-04-21:
I was trying to apply the patches using buildroot.

I'm no longer working on the project that triggered this PR, so I haven't kept up with the latest changes.

noob502, 2024-05-16:
> Hello bfoz,
> 
> the problem is carriage return/line feed related. If you are using the gnu patch, you could convert the patch files with: dos2unix file.patch After that, the gnu patch binary should be succeeding

Can you please explain more about this ?