VC-MIPI-modules/vc_mipi_nvidia #8: Support L4T 35.2.1 (Orin support)

pmliquify, 2023-05-15:


MartinHerren, 2023-05-15:
Had a draft PR open working on Orin NX with L4T 35.2.1 and 35.3.1 but it has been closed:
https://github.com/VC-MIPI-modules/vc_mipi_nvidia/pull/5

The PR was still draft as it didn't update the shell scripts or the README as we only used it integrated into our own BSP so we can use your scripts.

pmliquify, 2023-05-15:
Thanks a lot Martin, I had a look into it. Because of refactoring work on some scripts we won't use the PR as is, but used some code changes.

MartinHerren, 2023-06-02:
Hello,

for info i updated my Orin NX on DevKit device tree. It was lacking the polarity inversion on some lanes so only one camera worked before. Now both ports work fine. I could only test one port at a time as i have only one camera for evaluation purpose.

Tested working fine on a L4T 35.3.1 Orin NX on DevKit and custom carrier board with a VC IMX 265 camera.

https://github.com/MartinHerren/vc_mipi_nvidia/blob/orin_nx/src/devicetree/NV_DevKit_OrinNX/tegra234-p3768-camera-vc-mipi-cam.dtsi

Unfortunately still needs the `max_speed.sh` hack to work.

Best regards

pmliquify, 2023-07-10:
This issue will be closed. We support L4T 35.2.1 and L4T 35.3.1 with Release 0.14.0 on Xavier NX and AGX Xavier. See Issue #9 for more information on Jetson Orin Nano/NX support.

closed 2023-07-10 by pmliquify