VC-MIPI-modules/vc_mipi_nvidia #29: Gstreamer and OV9281 error

hobbeshunter, 2023-07-02:
Hi,

I have two OV9281 EK0033311 V1.0 from you which I wanted to connect to my Jetson Nano B01 dev board. However when I try to use them with gstreamer it fails. I've tried L4T versions 32.6.1 and 32.7.2.
After setting 
```sh
v4l2-ctl --set-fmt-video=pixelformat=RG10,width=1280,height=800
```
guvcview gives:
<details>
<summary>Show log</summary>

```
Opening in O_NONBLOCKING MODE 
libv4l2: error getting capabilities: Inappropriate ioctl for device
V4L2_CORE: ERROR opening V4L2 interface for /dev/v4l-subdev0
Opening in O_NONBLOCKING MODE 
libv4l2: error getting capabilities: Inappropriate ioctl for device
V4L2_CORE: ERROR opening V4L2 interface for /dev/v4l-subdev1
Opening in O_NONBLOCKING MODE 
libv4l2: error getting capabilities: Inappropriate ioctl for device
V4L2_CORE: ERROR opening V4L2 interface for /dev/v4l-subdev2
Opening in O_NONBLOCKING MODE 
libv4l2: error getting capabilities: Inappropriate ioctl for device
V4L2_CORE: ERROR opening V4L2 interface for /dev/v4l-subdev3
Opening in BLOCKING MODE
Available Sensor modes :
Resolution: 1280 x 800 ; Framerate = 120.599999; Analog Gain Range Min 0.000000, Max 12.000000, Exposure Range Min 1000, Max 1000000000

V4L2_CORE: Unable to find parent usb device.Opening in BLOCKING MODE
Available Sensor modes :
Resolution: 1280 x 800 ; Framerate = 120.599999; Analog Gain Range Min 0.000000, Max 12.000000, Exposure Range Min 1000, Max 1000000000

V4L2_CORE: Unable to find parent usb device.GUVCVIEW: version 2.0.5
GUVCVIEW: couldn't open /home/<user>/.config/guvcview2/video0 for read: No such file or directory
Opening in BLOCKING MODE
Available Sensor modes :
Resolution: 1280 x 800 ; Framerate = 120,599999; Analog Gain Range Min 0,000000, Max 12,000000, Exposure Range Min 1000, Max 1000000000

DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
DEFAULT no IOCTL called
V4L2_CORE: (UVCIOC_CTRL_MAP) Error: Invalid argument
V4L2_CORE: Error opening device /dev/video0: video capture not supported.
GUVCVIEW (1): Guvcview error
	 no video device found

(guvcview:22823): Gtk-WARNING **: 21:17:12.386: Theme parsing error: <data>:1:17: not a number

(guvcview:22823): Gtk-WARNING **: 21:17:12.387: Theme parsing error: <data>:1:31: Using Pango syntax for the font: style property is deprecated; please use CSS syntax

(guvcview:22823): Gtk-WARNING **: 21:17:12.387: Theme parsing error: <data>:1:17: not a number

(guvcview:22823): Gtk-WARNING **: 21:17:12.387: Theme parsing error: <data>:1:32: Using Pango syntax for the font: style property is deprecated; please use CSS syntax

(guvcview:22823): Gtk-WARNING **: 21:17:12.387: Theme parsing error: <data>:1:17: not a number

[... 161 lines cut ...]
  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

** Message: 21:27:49.643: <main:4670> iterating capture loop ....
ERROR on bus: by /GstPipeline:capture_native_pipeline/GstBin:cap_bin/GstV4l2Src:v4l2src0: Internal data stream error.
debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:capture_native_pipeline/GstBin:cap_bin/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
** Message: 21:27:49.703: <main:4680> Capture completed
** Message: 21:27:49.703: <main:4729> Camera application will now exit
```
</details>

Have I forgotten some setting?

pmliquify, 2023-07-11:
Hi hobbeshunter, guvcview and nvgstcapture-1.0 are not tested and therefore not supported. 
Please use gst-launch-1.0 to use gstreamer:
```
gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),framerate=20/1' ! autovideosink
```

hobbeshunter, 2023-07-22:
Thank you very much. That worked!

Follow up newby question: How do I use this with opencv?

I've tried 
```python
import cv2
import sys

gst_str = "nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),framerate=20/1' ! appsink"

cap = cv2.VideoCapture(gst_str, cv2.CAP_GSTREAMER)
if not cap.isOpened():
    print('Failed to open camera!')
    sys.exit()
while(True):
    _, img = cap.read()
    cv2.imshow("cam", img)
    key = cv2.waitKey(10)
```

without luck. OpenCV version is 3.2.0.

Thanks in advance!

hobbeshunter, 2023-07-31:
Any tipp on how to do it?

pmliquify, 2023-08-03:
Hi @hobbeshunter I am very busy currently. I try to support you from the 14th.

hobbeshunter, 2023-08-21:
Some more results from experiments:

When I try 

```
gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),framerate=20/1,format=NV12,width=1280,height=800' ! nvvidconv ! 'video/x-raw, format=GRAY8' ! autovideosink
```

I get following error:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 1280 x 800 FR = 120,599999 fps Duration = 8291874 ; Analog Gain range min 0,000000, max 12,000000; Exposure Range min 1000, max 1000000000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 0 
   Output Stream W = 1280 H = 800 
   seconds to Run    = 0 
   Frame Rate = 120,599999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
ERROR: from element /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.507162580
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
Setting pipeline to NULL ...
Freeing pipeline ...
```