VC-MIPI-modules/vc_mipi_nvidia #46: Infinite CSI Timeout in Trigger Mode

hesmar, 2023-08-02:
In trigger mode, the driver should dynamically adjust the CSI timeout to infinite. Otherwise the CSI interface will be resetted after this hardcoded time period. If the trigger event occures during this reset, the image will be dropped.
Increasing the hardcoded timeout in the driver seems not to be a valid option, because this makes the driver hang when the camera is stopped.

Tested on Jetson Nano, L4T 32.7.3/4

pmliquify, 2023-08-22:
Hello @hesmar, can you share with us your test implementation and a description of the test procedure?

hesmar, 2023-08-22:
Hi pmliquify,

you can test it with any V4L application. Just put the camera module in trigger mode (software or hardware). If you do not increase the timeout as described [here](https://github.com/VC-MIPI-modules/vc_mipi_nvidia#using-long-exposure-times-or-external-trigger-mode-with-long-waiting-times--5-seconds), the CSI interface will be resetted after five seconds. You will see timeout messages occur in the kernel log. If an image is transferred during this reset, this image will be dropped. Increasing the timeout helps in this case but leads to other problems. When stopping the stream and no image is sent from the camera (because no trigger signal occurred), the application hangs until the timeout exceeds.

For example, we have applications where a light barrier is used to trigger the camera. The duration between two triggers is unknown and can be relatively long (hours, days). Setting the timeout to such a high value is not an option because we would not be able to stop the stream again in a reasonable time. Keeping the value small leads to sporadic image drops.
Therefore it would be better to make this timeout dynamically adjustable. When enabling the trigger mode, the value should be set to a large value. When stopping the image stream, the value could be set to a smaller fixed value or depend on the exposure time.

pmliquify, 2023-08-23:
Hi @hesmar, thanks for your description. I hoped you had implemented the suggested dynamic timeout prototypically. We will have a look at this idea. Thanks again.

pmliquify, 2024-03-18:
This issue was fixed with Release 0.17.0 
https://github.com/VC-MIPI-modules/vc_mipi_nvidia/blob/v0.17.0/VERSION.md

closed 2024-03-18 by pmliquify