VC-MIPI-modules/vc_mipi_nvidia #48: XavierNX does not detect the camera

strijar, 2023-08-15:
Use NV_DevKit_XavierNX. DTS from Auvidea_J20_AGXXavier is erroneously selected in scripts. But even if you turn on the right DTS, it still doesn't work.

vc@XavierNX:~$ sudo dmesg | grep -i i2c
[sudo] password for vc: 
[    2.911452] i2c /dev entries driver
[    2.914621] tegra-i2c 3160000.i2c: Adding to iommu group 2
[    2.920412] tegra-i2c c240000.i2c: Adding to iommu group 2
[    2.926083] tegra-i2c 3180000.i2c: Adding to iommu group 2
[    2.931133] tegra-i2c 3190000.i2c: Adding to iommu group 2
[    2.936469] tegra-i2c 31b0000.i2c: Adding to iommu group 2
[    2.942184] tegra-i2c 31c0000.i2c: Adding to iommu group 2
[    2.947341] tegra-i2c c250000.i2c: Adding to iommu group 2
[    5.508356] tegra-i2c 31e0000.i2c: Adding to iommu group 2
[    7.914239] vc_mipi 9-001a: vc_mod_setup(): Unable to get module I2C client for address 0x10
[    7.914714] i2c i2c-2: Added multiplexed i2c bus 9
[   10.314191] vc_mipi 10-001a: vc_mod_setup(): Unable to get module I2C client for address 0x10
[   10.314692] i2c i2c-2: Added multiplexed i2c bus 10
[   10.314805] i2c-mux-gpio cam_i2cmux: 2 port mux on 3180000.i2c adapter

vc@XavierNX:~$ sudo dmesg | grep -i mipi
[    0.006517] DTS File Name: /home/master/src/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t19x/jakku/kernel-dts/tegra194-p3668-0001-p3509-0000.dts
[    5.511250] vc_mipi 9-001a: vc_probe(): Probing UNIVERSAL VC MIPI Driver (v0.14.1)
[    7.914239] vc_mipi 9-001a: vc_mod_setup(): Unable to get module I2C client for address 0x10
[    7.914447] vc_mipi 9-001a: vc_probe(): Error in vc_core_init!
[    7.915673] vc_mipi 10-001a: vc_probe(): Probing UNIVERSAL VC MIPI Driver (v0.14.1)
[   10.314191] vc_mipi 10-001a: vc_mod_setup(): Unable to get module I2C client for address 0x10
[   10.314427] vc_mipi 10-001a: vc_probe(): Error in vc_core_init!

bazo80, 2023-08-23:
Hello strijar,

have you used the 
`sudo ./flash.sh -d`
command for flashing the device-tree? 
If so, than you have to comment out the FTD line in your /boot/extlinux/extlinux.conf file since you are running on a Jetpack 5.

bazo80, 2023-08-24:
Could you give us a little more information about your procedure? Which script calls did you use, which sensor(s) have you attached and so on. Maybe we can reproduce it. Either the problem is related to the device tree include file issue #44 or we have an unknown error of which we are not aware yet.
There is a branch 19-check-all-downloaded-files-with-md5-sum_master, which will deal with the issue #44 
You could use this one and have a test with your setup, maybe it already solves your problem.

AhmedL99, 2024-07-10:
We got the same issue with NV Xavier NX, the i2c address 0x1a is busy and we got the same dimes log 
`
[    7.959841] vc_mipi 10-001a: vc_mod_setup(): Unable to get module I2C client for address 0x10
[    7.960070] vc_mipi 10-001a: vc_probe(): Error in vc_core_init!
`