VC-MIPI-modules/vc_mipi_nvidia #49: Issues with VC-MIPI Driver Version 0.14 on Jetson Xavier NX with Auvidea JNX30 Carrier Board

HBanjak, 2023-08-17:
I have observed several problems while using VC-MIPI Driver version 0.14 on my Jetson Xavier NX with the Auvidea JNX30 Carrier Board to test it with the IMX183 Sensor.

1. **Device Tree Selection Issue:**
During the flashing process, the incorrect Device Tree was displayed. The message indicated the usage of a device tree camera file from the path: ../Auvidea_J20_AGXXavier/..., while I had selected XavierNX and Auvidea_JNX30 as shown below:
[image]

2. **Pixelformat GREY Error:**
After successfully flashing the Jetson, I installed V4L (Video for Linux) and attempted to select the Pixelformat GREY, which I had used previously. However, I encountered an error message stating that this Pixelformat was invalid. Only the RGGB 8 bit format was available in the list-formats output. (I had configured this Pixelformat in the devicetree.)
[image]
[image]

3. **Streaming and Test Image Issue:**
I also tried to create a test image using a V4L command, but this process failed. Whenever I attempt to start streaming, the system tries to capture an image every 10 seconds (as set in vi5_fops.c), but it fails to do so. The command I used is:
v4l2-ctl -d /dev/video0 --set-ctrl bypass_mode=0 --stream-mmap --stream-count=1 --stream-to=test_5440x3648.raw
[image]

pmliquify, 2023-08-22:
Hi @HBanjak, thanks for your feedback. Your first Issue will be solved with the next Release and will be tracked in #44 .Your second and third issue has to investigated by us.

bazo80, 2023-09-20:
Hello @HBanjak, 

the pixelformat grey error can be fixed with multiple modes in the device-tree. In former v4l-utils versions, the string "GREY" has been converted and given to the kernel via ioctl. If it was defined, the kernel could successfully return the ioctl and it was fine. Now the newer v4l-utils will enumerate the available formats and if the format name corresponds to the given name ('RGGB' in your example), the ioctl set command will succeed and the format is set. "GREY" is still defined, but not used in the created array of the v4l2-ctl binary. 

What you can do is placing an additional mode like this way:

```
mode0 {
	num_lanes                = NUM_LANES;
	tegra_sinterface         = "serial_a";
	...

#if LINUX_VERSION < 500
	pixel_t                  = "bayer_rggb";
#else
	mode_type                = "bayer";
	pixel_phase              = "rggb";
	csi_pixel_bit_depth      = "8";
#endif
	min_gain_val             = "0";         // mdB
	max_gain_val             = "27000";     // mdB
	...
};
mode1 {
	num_lanes                = NUM_LANES;
	tegra_sinterface         = "serial_a";
	...

#if LINUX_VERSION < 500
	pixel_t                  = "bayer_rggb";
#else
	mode_type                = "bayer";
	pixel_phase              = "rggb";
	csi_pixel_bit_depth      = "10";
#endif
	min_gain_val             = "0";         // mdB
	max_gain_val             = "27000";     // mdB
	...
};
```

This will add another entry 'RG10' in your v4l2-ctl --list-formats call and so it becomes switchable again.

bazo80, 2023-10-24:
Hello @HBanjak 

the third problem you mentioned is related to the frequencies in Jetpack5 (35.x.y)
There you have to call the max_speed.sh script

   ```
     $ sudo ./max_speed.sh
   ```
should solve the problem.

HBanjak, 2023-10-24:
Hello @bazo80 

Excellent, thanks for your response!