VC-MIPI-modules/vc_mipi_nvidia #58: After flashing Jetson Orin Nano, GUI not showing.

xaprier, 2023-11-07:
After flashing with using branch '9-support-for-jetson-orin-nano', GUI or display not working.

- minicom working and looks like it's flashed successfully.
- When using 'ACPI' on BIOS rather than 'Device Tree' option, GUI and display came but usb ports don't work.
- In the script seems like everything is successful.
- Selected camera: IMX568, but camera not connected to device.(Should it be connected?)

bazo80, 2023-12-04:
Hello @xaprier ,

are there any dmesg printings related to nvidia: (missing symbol or something like that)? I have faced this problem with OrinNX and HDMI. There seems to be an issue with Orin and display driver in general when using a custom kernel. https://docs.nvidia.com/jetson/archives/r35.3.1/DeveloperGuide/text/SD/Kernel/KernelCustomization.html
They are mentioning a black screen fix by performing:
```
sudo depmod -a
```
But that did not solve the problem for OrinNX on my Carrier-Board, so I compiled the display driver in addition to the kernel. After that, it loaded the display drivers and the GUI was available. We are about to implement this build step in our pipeline, too. And then we have to test it with OrinNano-Devkit and DisplayPort.

If you are seeing such dmesg printings stating that the nvidia display driver has not been loaded and also the depmod call did not succeed, then you could try to build the display driver nvidia.ko, nvidia-modeset, nvidia-drm on your own. There is the nvidia_kernel_display_driver_source.tbz2 which you can extract and cross compile. After that you should copy the kernel objects into the rootfs. 
If this is not working, we have to take a closer look.

Btw. the IMX sensor don't need to be connected for this.

xaprier, 2023-12-05:
Hello @bazo80,

Thank you for answering first of all.

I looked over dmesg output but nothing interesting showed up. I will attach the output file.

I tried ```sudo depmod -a``` with minicom and rebooted, but nothing changed. When I looked into the [Nvidia DOCS](https://docs.nvidia.com/jetson/archives/r35.4.1/DeveloperGuide/text/SD/Kernel/KernelCustomization.html#to-build-the-real-time-kernel), I saw the same operation as you said.

When I try to compile it, I encountered a problem with build; 
```
make[1]: Entering directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/kernel-open'
make[2]: Entering directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10'
make[3]: Entering directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel_out'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10/Makefile:675: include/config/auto.conf: No such file or directory
make[3]: *** [/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10/Makefile:750: include/config/auto.conf] Error 1
```

Used build command;
```sh
make \
      modules \
      -j8 \
      SYSSRC=/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10 \
      SYSOUT=/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel_out \
      CC=/home/burak/vc_mipi_nvidia/build/toolchain/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc \
      LD=/home/burak/vc_mipi_nvidia/build/toolchain/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-ld.bfd \
      AR=/home/burak/vc_mipi_nvidia/build/toolchain/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-ar \
      CXX=/home/burak/vc_mipi_nvidia/build/toolchain/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-g++ \
      OBJCOPY=/home/burak/vc_mipi_nvidia/build/toolchain/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-objcopy \
      TARGET_ARCH=aarch64 \
      ARCH=arm64
```

When I tried to what build says, I got encountered with this error;
```
 [ nvidia-modeset    ]  CC           _out/Linux_aarch64/g_nvid_string.c
 [ nvidia-modeset    ]  LD           _out/Linux_aarch64/nv-modeset-kernel.o
make[1]: Leaving directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/src/nvidia-modeset'
cd kernel-open/nvidia-modeset/ && ln -sf ../../src/nvidia-modeset/_out/Linux_aarch64/nv-modeset-kernel.o nv-modeset-kernel.o_binary
make -C kernel-open modules
make[1]: Entering directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/kernel-open'
make[2]: Entering directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10'
make[3]: Entering directory '/home/burak/Desktop/out'
scripts/Makefile.lib:8: 'always' is deprecated. Please use 'always-y' instead
The Module.symvers file is missing, or does not contain any
symbols exported from the kernel. This could cause the NVIDIA
kernel modules to be built against a configuration that does
not accurately reflect the actual target kernel.
The Module.symvers file check can be disabled by setting the
environment variable IGNORE_MISSING_MODULE_SYMVERS to 1.
make[4]: *** [/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/kernel-open/Kbuild:234: module_symvers_sanity_check] Error 1
make[3]: *** [/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10/Makefile:1854: /home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/kernel-open] Error 2
make[3]: Leaving directory '/home/burak/Desktop/out'
make[2]: *** [Makefile:213: __sub-make] Error 2
make[2]: Leaving directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/kernel/kernel-5.10'
make[1]: *** [Makefile:82: modules] Error 2
make[1]: Leaving directory '/home/burak/vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/source/public/NVIDIA-kernel-module-source-TempVersion/kernel-open'
make: *** [Makefile:50: modules] Error 2
```

Can you share the build steps on display driver with details? 
[dmesg-output.txt](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/13559920/dmesg-output.txt)

xaprier, 2023-12-06:
Hello again @bazo80,

I tried what you said today, and no results...

Build was successfully, and I tried 2 ways to try it.
- Copy the *.ko modules into /usr/lib/modules/5.10.104-tegra/extra/opensrc-disp folder in Jetson device directly. That didn't work.
- Copy the *.ko modules into vc_mipi_nvidia/build/Xavier_35.3.1/Linux_for_Tegra/rootfs/usr/lib/modules/5.10.104-tegra/extra/opensrc-disp directh and then used flash.sh script also didn't work.

### Here is the log outputs
#### Xorg
```
[    21.963] (--) Log file renamed from "/var/log/Xorg.pid-1542.log" to "/var/log/Xorg.0.log"
[    21.965] 
X.Org X Server 1.20.13
X Protocol Version 11, Revision 0
[    21.965] Build Operating System: linux Ubuntu
[    21.965] Current Operating System: Linux nvidia 5.10.104-tegra #1 SMP PREEMPT Tue Dec 5 16:31:46 +03 2023 aarch64
[    21.965] Kernel command line: root=PARTUUID=5b8eb85c-540d-4c86-9244-ca86be02eb1c rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 console=ttyAMA0,115200 firmware_class.path=/etc/firmware 
[    21.965] Build Date: 07 February 2023  12:48:13PM
[    21.965] xorg-server 2:1.20.13-1ubuntu1~20.04.6 (For technical support please see http://www.ubuntu.com/support) 
[    21.965] Current version of pixman: 0.38.4
[    21.965]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    21.965] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    21.965] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Sep  8 09:58:22 2022
[    21.967] (==) Using config file: "/etc/X11/xorg.conf"
[    21.967] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    21.967] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    21.968] (==) No Layout section.  Using the first Screen section.
[    21.968] (==) No screen section available. Using defaults.
[    21.968] (**) |-->Screen "Default Screen Section" (0)
[    21.968] (**) |   |-->Monitor "<default monitor>"
[    21.969] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    21.969] (**) |   |-->Device "Tegra0"
[    21.969] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    21.969] (==) Automatically adding devices
[    21.969] (==) Automatically enabling devices
[    21.969] (==) Automatically adding GPU devices
[    21.969] (==) Automatically binding GPU devices
[    21.969] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    21.970] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    21.970]    Entry deleted from font path.
[    21.970] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    21.970]    Entry deleted from font path.
[    21.970] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    21.970]    Entry deleted from font path.
[    21.970] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    21.970]    Entry deleted from font path.
[    21.970] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    21.970]    Entry deleted from font path.
[    21.970] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        /usr/share/fonts/X11/Type1,
        built-ins
[    21.970] (==) ModulePath set to "/usr/lib/xorg/modules"
[    21.970] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    21.970] (II) Loader magic: 0xaaaadf1aa010
[    21.970] (II) Module ABI versions:
[    21.970]    X.Org ANSI C Emulation: 0.4
[    21.970]    X.Org Video Driver: 24.1
[    21.970]    X.Org XInput driver : 24.1
[    21.970]    X.Org Server Extension : 10.0
[    21.971] (++) using VT number 2

[    21.974] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_31
[... 1123 lines cut ...]
[   21.427371] l4tbr0: port 1(rndis0) entered blocking state
[   21.427375] l4tbr0: port 1(rndis0) entered forwarding state
[   21.438815] IPv6: ADDRCONF(NETDEV_CHANGE): l4tbr0: link becomes ready
[   21.548190] nvgpu: 17000000.ga10b                  gk20a_scale_init:539  [INFO]  enabled scaling for GPU

[   21.693208] zram: Added device: zram0
[   21.694558] zram: Added device: zram1
[   21.695046] zram: Added device: zram2
[   21.695408] zram: Added device: zram3
[   21.695627] zram: Added device: zram4
[   21.695772] zram: Added device: zram5
[   21.749677] zram0: detected capacity change from 0 to 641110016
[   21.778964] Adding 626080k swap on /dev/zram0.  Priority:5 extents:1 across:626080k SS
[   21.783044] zram1: detected capacity change from 0 to 641110016
[   21.818775] Adding 626080k swap on /dev/zram1.  Priority:5 extents:1 across:626080k SS
[   21.819939] zram2: detected capacity change from 0 to 641110016
[   21.846755] Adding 626080k swap on /dev/zram2.  Priority:5 extents:1 across:626080k SS
[   21.847969] zram3: detected capacity change from 0 to 641110016
[   21.882786] Adding 626080k swap on /dev/zram3.  Priority:5 extents:1 across:626080k SS
[   21.883933] zram4: detected capacity change from 0 to 641110016
[   21.914775] Adding 626080k swap on /dev/zram4.  Priority:5 extents:1 across:626080k SS
[   21.915924] zram5: detected capacity change from 0 to 641110016
[   21.954762] Adding 626080k swap on /dev/zram5.  Priority:5 extents:1 across:626080k SS
[   22.491019] fuse: init (API version 7.32)
```

xaprier, 2023-12-06:
After all operation done, I forgot to try ```sh sudo depmod -a```, when I tried, it worked... Now display is seems working.

closed 2023-12-06 by xaprier