VC-MIPI-modules/vc_mipi_nvidia #64: Can not get image or black image with IMX568c

musafzenbilci, 2024-02-05:
Here are the steps I followed:
In x86 ubuntu 20.08 host;
1- Clone your driver repo
2- Switch to `jetson-orin-nano` branch
3- Run `./quickstart.sh`
4- Edit the device tree for imx568c as 
```
/*
 * Copyright (c) 2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano on NVIDIA Jetson Orin Nano Developer Kit
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_a => CSI Port 0 => VI Stream 0)
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       2   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "4" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
//  this repository https://github.com/VC-MIPI-modules/vc_mipi_nvidia#gstreamer-support
// ------------------------------------------------------------------------------------------------

#if VC_MIPI_LANES == 1
    #define BUS_WIDTH 1
    #define NUM_LANES "1" 
    #define NUM_CSI_LANES 2
	#define PIX_CLK_HZ "150000000"
[... 524 lines cut ...]
				1: 1 (0x1)
                override_enable 0x009a2065 (intmenu): min=0 max=1 default=0 value=0
				0: 0 (0x0)
				1: 1 (0x1)
                   height_align 0x009a2066 (int)    : min=1 max=16 step=1 default=1 value=1
                     size_align 0x009a2067 (intmenu): min=0 max=2 default=0 value=0
				0: 1 (0x1)
				1: 65536 (0x10000)
				2: 131072 (0x20000)
               write_isp_format 0x009a2068 (int)    : min=1 max=1 step=1 default=1 value=1
       sensor_signal_properties 0x009a2069 (u32)    : min=0 max=4294967295 step=1 default=0 [30][18] flags=read-only, has-payload
        sensor_image_properties 0x009a206a (u32)    : min=0 max=4294967295 step=1 default=0 [30][18] flags=read-only, has-payload
      sensor_control_properties 0x009a206b (u32)    : min=0 max=4294967295 step=1 default=0 [30][36] flags=read-only, has-payload
              sensor_dv_timings 0x009a206c (u32)    : min=0 max=4294967295 step=1 default=0 [30][16] flags=read-only, has-payload
               low_latency_mode 0x009a206d (bool)   : default=0 value=0
               preferred_stride 0x009a206e (int)    : min=0 max=65535 step=1 default=0 value=0
                   sensor_modes 0x009a2082 (int)    : min=0 max=30 step=1 default=30 value=1 flags=read-only
```
                   
7- Then I am trying different things to capture image but all of them either gets error or returns black image.

Possible problems might be:
 - Jetson 35.3.1 Driver files are outdated in your repo but I downloaded them manually and still does not work
 - My device tree file has something wrong but then would it be appeared under /dev
 - Something outdated in the bash scripts

AOSMarbacher, 2024-02-15:
Hello
We use the Sony IMX586 via MIPI /CSI on an NVIDIA Jetson Xavier NX. We can stream images, but the images are displayed two and a half times. (see attached image).
[image]

We use the following GStreamer command:
“gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! ‘video/x-raw(memory:NVMM),width=2464,height=2048,format=(string)NV12,framerate=(fraction)30/1’ ! nvvidconv ! queue ! autovideosink”

Does anyone have any idea what we have configured wrong?

AOSMarbacher, 2024-02-16:
Yes we used the nvgstcapture-1.0 but with the same  two and a half times image-
Yes we integraded the values to the device tree file (see attached file).
Do you now have images from the camera on your system?
[tegra194-camera-vc-mipi-cam.txt](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/14310972/tegra194-camera-vc-mipi-cam.txt)

AOSMarbacher, 2024-02-16:
And what did you do that it works now?

AOSMarbacher, 2024-02-19:
On the OS setup what L4T version do you use?
[image]

Do you use the VC Module with 2 or 4 lines?

AOSMarbacher, 2024-02-19:
I found the solution for our problem. I changed the "pix_clk_hz" to "#define PIX_CLK_HZ "300000000"" that the same value in your device tree file. And now we have a proper image.
Where did you find that value?

musafzenbilci, 2024-02-19:
I did not changed anything like that actually. Only changed the device tree for imx568c specifics for gstreamer and lane,metadata number etc. Also I installed on Jetson Orin Nano with Jetson Linux 35.3.1. That might be the difference maybe.

bazo80, 2024-02-20:
Hello @musafzenbilci, hello @AOSMarbacher 

we made a differentiation of the PIX_CLK_HZ depending on the lane number. For the time being this done for OrinNX/Orin Nano only.
For systems like Xavier NX, there is still the old frequency of 74250000 Hz configured (L4T35.1.0 and later). In order to stay compatible with former L4T versions (32.7.3 and earlier), which the Xavier NX is supporting too, there is a different switch.
Maybe we will refactor the dtsi files for a more convenient handling for customers, but this is connected with a lot of test effort.

@AOSMarbacher at the end of the README.md there is the max_speed.sh script mentioned which should be called for L4T 35.1.0 and above. If you are using the default frequency of 74250000 Hz and run the max_speed.sh as superuser after the boot process, then the sensor should work properly.

AOSMarbacher, 2024-02-20:
Hello @bazo80 
Thank you for the information.
Where do we find the max_speed.sh script? We could not find it on the NVIDIA Xavier NX.

musafzenbilci, 2024-02-20:
it is under `target` folder in the repo.

AOSMarbacher, 2024-02-20:
thank you that was the right hint

bazo80, 2024-02-20:
@musafzenbilci yes, I just brought the bazo_pre_3541 branch online. This branch won't be included in the 0.17.0 Release, but in the next one.

musafzenbilci, 2024-02-21:
Thanks, I will try that next week and will get back to you.

musafzenbilci, 2024-02-27:
hey @bazo80 , i tried your `35.4.1` branch. 
At first, I got the following errors during  run of `quickstart.sh` after the config part. I made the config same as I did before

```
E: The repository 'file:/var/cuda-repo-cross-aarch64-ubuntu2004-11-4-local  Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'file:/var/cuda-repo-ubuntu2004-11-4-local  Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Then I handled this error by removing some records about cuda from apt releases on my local. After that it run without error and ended in 3 hours succesfully.  Then I opened my target device which is a Jetson Orin Nano Devkit NVMM 8GB and run `sudo maxspeed.sh`. I can see the device from the output of `v4l2-ctl --list-devices`:

```
vi-output, vc_mipi 9-001a (platform:tegra-capture-vi:1):
        /dev/video0
```

then I tried running `nvgstcapture-1.0` and some gstreamer pipelines I knew working with 35.3.1 and all of them returned this error and only black feed.

```
`nvbuf_utils: dmabuf_fd -1 mapped entry NOT found`
```

After all these, I did the driver installation again and it gave the same error again.

bazo80, 2024-02-27:
Hello @musafzenbilci ,

please be aware, that the max_speed.sh script has been refactored. It is needing now a parameter. In that case -m
<pre>
sudo ./max_speed.sh <b>-m</b>
</pre>

On this pre_release branch it has not been documented, but on the current develop and the bazo_integ_0170 branch it is decribed in the README.md 
Please have a try.

And the other problem regarding the broken cuda install. I haven't faced this error yet.

musafzenbilci, 2024-02-28:
Hi @bazo80,

I run the `sudo max_speed.sh -m` as you said on the target but nothing changed. 

When I run `nvgstcapture-1.0`, it logs the error and returns only a blackscreen. Should I use other branches `develop` or `bazo_integ_0170`?

```
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadExecute:694 NvBufSurfaceFromFd Failed.
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:247 (propagating)
```

musafzenbilci, 2024-02-28:
It is not supported as attachment so I am sharing its content here:

````

/*
 * Copyright (c) 2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY orientation
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano on NVIDIA Jetson Orin Nano Developer Kit
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_a => CSI Port 0 => VI Stream 0)
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       2   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "4" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
//  this repository https://github.com/VC-MIPI-modules/vc_mipi_nvidia#gstreamer-support
// ------------------------------------------------------------------------------------------------

#if VC_MIPI_LANES == 1
    #define BUS_WIDTH 1
    #define NUM_LANES "1" 
    #define NUM_CSI_LANES 2
	#define PIX_CLK_HZ "150000000"
#endif
#if VC_MIPI_LANES == 2
    #define BUS_WIDTH 2
[... 396 lines cut ...]
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;
#if VC_MIPI_MANUFACTURER == 1
			vc_mipi@1a {
#else
            		vc_mipi@60 {
#endif
				reset-gpios = <&tegra_main_gpio CAM1_PWDN GPIO_ACTIVE_HIGH>;
			};
		};
#endif
	};

	gpio@2200000 {
		camera-control-output-low {
			gpio-hog;
			output-low;
			gpios = <CAM0_PWDN 0 CAM1_PWDN 0>;
			label = "cam0-pwdn","cam1-pwdn";
		};
	};
};

````

bazo80, 2024-02-28:
Hello @musafzenbilci ,

this dtsi seems to be a little bit outdated. The port index should be 1, the lane_polarity should be 6 and the tegra_sinterface should be "serial_b" (nvargus is looking at this one), for the first sensor.
<pre>
/*
 * Copyright (c) 2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano on NVIDIA Jetson Orin Nano Developer Kit
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_b => CSI Port 1 => VI Stream 0)
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX462, IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       2   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "4" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
//  this repository https://github.com/VC-MIPI-modules/vc_mipi_nvidia#gstreamer-support
// ------------------------------------------------------------------------------------------------

#if VC_MIPI_LANES == 1
        #define BUS_WIDTH 1
        #define NUM_LANES "1" 
        #define NUM_CSI_LANES 2
        #define PIX_CLK_HZ "150000000"
#endif
#if VC_MIPI_LANES == 2
        #define BUS_WIDTH 2
[... 397 lines cut ...]
                        #address-cells = <1>;
                        #size-cells = <0>;
#if VC_MIPI_MANUFACTURER == 1
                        vc_mipi@1a {
#else
                        vc_mipi@60 {
#endif
                                reset-gpios = <&tegra_main_gpio CAM1_PWDN GPIO_ACTIVE_HIGH>;
                        };
                };
#endif
        };

        gpio@2200000 {
                camera-control-output-low {
                        gpio-hog;
                        output-low;
                        gpios = <CAM0_PWDN 0 CAM1_PWDN 0>;
                        label = "cam0-pwdn","cam1-pwdn";
                };
        };
};
</pre>

please have a try with this one.

musafzenbilci, 2024-02-29:
@bazo80  Driver installation failed with this.Are you sure this is syntatically right? There are some lines I am not sure correct like

- empty `#include ` at line 18
- empty assignment `num-channels = ;` at line 86 and 115
- empty assignment `bus-width = ;` at line 95 and 105 and 128 and 151
and so on. I did not scan all of the file. I will restore these assignments to the previous file and try again.

bazo80, 2024-02-29:
Hello @musafzenbilci ,

yes you are right. For some reason, the "\<pre>" tag is cutting out some parts.

I attached the dtsi-file. The file name must be renamed to tegra234-camera-vc-mipi-cam.dtsi.
[new_orinnano.dtsi.txt](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/14446090/new_orinnano.dtsi.txt)

I hope the file is complete now :)
Basically the points are the first port index, the first lane polarity and the first tegra_sinterface.

musafzenbilci, 2024-02-29:
Alright, it is working now. I can see the feed of camera. That is really great work. You helped me a lot @bazo80  I appreciate it. Also what is the future of this branch? Is this gonna be merged or stay in branch?

bazo80, 2024-02-29:
You are welcome.
You mean the L4T 35.4.1?
According to our flight plan, we will integrate that into the next release (0.18.0)

Best regards