VC-MIPI-modules/vc_mipi_nvidia #66: nvarguscamerasrc fails

nirushka, 2024-02-19:
Hi,

I am using IMX226, with L4T 35.1.0

I get this error with the current pipeline: gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),width=3904,height=3000,framerate=20/1' ! nvvidconv ! video/x-raw, width=1024, height=768, format=I420 ! xvimagesink

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
(Argus) Error BadParameter:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function initialize(), line 89)
(Argus) Error BadParameter:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function create(), line 44)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadInitialize:320 Failed to create FrameConsumer
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:241 (propagating)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, waitRunning:203 Invalid thread state 3
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:806 (propagating)
WARNING: from element /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0: Pipeline construction is invalid, please add queues.
Additional debug info:
gstbasesink.c(1209): gst_base_sink_query_latency (): /GstPipeline:pipeline0/GstXvImageSink:xvimagesink0:
Not enough buffering available for  the processing deadline of 0:00:00.015000000, add enough queues to buffer  0:00:00.015000000 additional data. Shortening processing latency to 0:00:00.000000000.
Got EOS from element "pipeline0".
Execution ended after 0:00:00.330253389
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

with v4l2-ctl command: v4l2-ctl -d /dev/video0 --verbose --set-fmt-video=width=3904,height=3000,pixelformat=RG10 --stream-mmap
everything works fine.

bazo80, 2024-02-20:
Hello @nirushka 

have you tried to insert an extra queue before the sink?
gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),width=3904,height=3000,framerate=20/1' ! nvvidconv ! video/x-raw, width=1024, height=768, format=I420 ! <b>queue !</b> xvimagesink

I've just tested this construct with L4T35.3.1 and IMX568. With the first try I've got this warning too. And with the additional queue, I got rid of the warning. It might be, that this warning could be the reason, since you're running an older L4T.

nirushka, 2024-02-20:
Hi, 
This is not a warning... it's an error.
Same problem with a queue.

and even when I try: gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! fakesink
I get this error.

nirushka, 2024-02-28:
Hi, any news about this topic?

This is my v4l-ctl --all output: 

Driver Info:
	Driver name      : tegra-video
	Card type        : vi-output, vc_mipi 9-001a
	Bus info         : platform:tegra-capture-vi:0
	Driver version   : 5.10.104
	Capabilities     : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : tegra-camrtc-ca
	Model            : NVIDIA Tegra Video Input Device
	Serial           : 
	Bus info         : 
	Media version    : 5.10.104
	Hardware revision: 0x00000003 (3)
	Driver version   : 5.10.104
Interface Info:
	ID               : 0x0300000b
	Type             : V4L Video
Entity Info:
	ID               : 0x00000009 (9)
	Name             : vi-output, vc_mipi 9-001a
	Function         : V4L2 I/O
	Pad 0x0100000a   : 0: Sink
	  Link 0x0200000f: from remote pad 0x1000003 of entity '13e10000.host1x:nvcsi@15a00000-': Data, Enabled
Priority: 2
**Video input : 0 (Camera 0: no power)**
Format Video Capture:
	Width/Height      : 3904/3000
	Pixel Format      : 'GB10' (10-bit Bayer GBGB/RGRG)
	Field             : None
	Bytes per Line    : 7808
	Size Image        : 23424000
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Selection Video Capture: crop, Left 0, Top 0, Width 3904, Height 3000, Flags: 
Selection Video Output: crop, Left 0, Top 0, Width 3904, Height 3000, Flags: 

Camera Controls

                     group_hold 0x009a2003 (bool)   : default=0 value=0 flags=execute-on-write
                    sensor_mode 0x009a2008 (int64)  : min=0 max=1 step=1 default=0 value=0 flags=slider
                           gain 0x009a2009 (int64)  : min=0 max=27000 step=100 default=0 value=0 flags=slider
                       exposure 0x009a200a (int64)  : min=1 max=1951310 step=1 default=10000 value=1 flags=slider
                     frame_rate 0x009a200b (int64)  : min=0 max=60897 step=1 default=0 value=0 flags=slider
                   trigger_mode 0x009a200e (int)    : min=0 max=7 step=1 default=0 value=0
                        io_mode 0x009a200f (int)    : min=0 max=5 step=1 default=0 value=0
                    black_level 0x009a2010 (int)    : min=0 max=100000 step=1 default=0 value=0
                 single_trigger 0x009a2011 (button) : flags=write-only, execute-on-write
           sensor_configuration 0x009a2032 (u32)    : min=0 max=4294967295 step=1 default=0 [22] flags=read-only, volatile, has-payload
         sensor_mode_i2c_packet 0x009a2033 (u32)    : min=0 max=4294967295 step=1 default=0 [1026] flags=read-only, volatile, has-payload
      sensor_control_i2c_packet 0x009a2034 (u32)    : min=0 max=4294967295 step=1 default=0 [1026] flags=read-only, volatile, has-payload
                    bypass_mode 0x009a2064 (intmenu): min=0 max=1 default=0 value=0
				0: 0 (0x0)
				1: 1 (0x1)
                override_enable 0x009a2065 (intmenu): min=0 max=1 default=0 value=0
				0: 0 (0x0)
[... 511 lines cut ...]
                        status = "okay";
                        reg = <1>;
                        #address-cells = <1>;
                        #size-cells = <0>;
#if VC_MIPI_MANUFACTURER == 1
                        vc_mipi@1a {
#else
                        vc_mipi@60 {
#endif
                                status = "okay";
                                reset-gpios = <&tegra_main_gpio CAM1_PWDN GPIO_ACTIVE_HIGH>;
                        };
                };
#endif
        };

        gpio@2200000 {
                camera-control-output-low {
                        gpio-hog;
                        output-low;
                        gpios = <CAM0_PWDN 0 CAM1_PWDN 0>;
                        label = "cam0-pwdn","cam1-pwdn";
                };
        };
};

bazo80, 2024-02-28:
Hello @nirushka ,

you have configured 2 lanes
#define VC_MIPI_LANES 2 // 1, 2 Lanes
but your frame rate is:
max_framerate            = "43600"; 
default_framerate        = "43600";

I would try to change it either to 2 lanes and 21800 mHz or
4 lanes and 43600 mHz

I guess, we should extend the comment for this device to // 1, 2, 4 lanes

nirushka, 2024-03-05:
I found out that this error occurs only if I connect via **ssh -X**.
I tried to export display=:0 as many forums say but it didn't work. 
Any thoughts?