VC-MIPI-modules/vc_mipi_nvidia #75: Do I need to change dtsi for Camera Triggering

musafzenbilci, 2024-03-11:
You can also see the [discussion](https://forums.developer.nvidia.com/t/cant-use-cam0-pwdn-on-jetson-orin-nano-devkit-over-script/285388) in NVIDIA Forum.

I am using the IMX568c camera on Jetson Linux 35.4.1 within Jetson Orin Nano. I installed the driver from the `bazo_pre_3541` pre-release branch with this dtsi file 
[new_orinnano.dtsi.txt](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/14559726/new_orinnano.dtsi.txt)
 and the camera worked as expected. 

Now I needed to use the trigger feature of the camera but could not find a way to work it out. Created a discussion under NVIDIA Forum since the pinouts related to them and **at the end, he told me to remove the assignment of `CAM0_PWDN` in the device tree. Should I really make that kind of change to use the trigger feature of yours?**

I am not experienced in using camera triggers so I want to explore both the hardware and software triggers. I read the trigger documentation in this repo and opened the single trigger mode with `v4l2-ctl -c trigger_mode=4` and then tried triggering with `v4l2-ctl -c single_trigger=1` but I did not notice any additional frame captured from the camera feed. 

To test the trigger, I reduced the fps to 1/10 and ran `v4l2-ctl -c single_trigger=1`  to see any frame captured with it but nothing changed in the camera feed. Here is the pipeline I used for testing: 

```
gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM),width=2464, height=2048, format=NV12' ! nvvidconv ! videorate ! "video/x-raw,framerate=1/10" ! nvvidconv ! nvegltransform ! nveglglessink -e
```

noob502, 2024-05-07:
Did you build the Jetson Orin Nano image or use any pre-built image?

musafzenbilci, 2024-05-07:
pre-built image