VC-MIPI-modules/vc_mipi_nvidia #83: error: ‘tRet’ may be used uninitialized in this function [-Werror=maybe-uninitialized]

RoyAmoyal, 2024-03-24:
Hey, I am trying to install the drivers for our imx327-c camera but I am getting the next errors:

```
/home/copterpixroy/Desktop/jetson_camera_imx/vc_mipi_nvidia/build/Nano_32.5.2/Linux_for_Tegra/source/public/kernel/nvidia/drivers/media/i2c/vc_mipi_core.c: In function ‘vc_core_get_mode’:
/home/copterpixroy/Desktop/jetson_camera_imx/vc_mipi_nvidia/build/Nano_32.5.2/Linux_for_Tegra/source/public/kernel/nvidia/drivers/media/i2c/vc_mipi_core.c:648:16: error: ‘tRet’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
         return tRet;
                ^~~~
cc1: all warnings being treated as errors
/home/copterpixroy/Desktop/jetson_camera_imx/vc_mipi_nvidia/build/Nano_32.5.2/Linux_for_Tegra/source/public/kernel/kernel-4.9/scripts/Makefile.build:335: recipe for target 'drivers/media/i2c/vc_mipi_core.o' failed
make[4]: *** [drivers/media/i2c/vc_mipi_core.o] Error 1
/home/copterpixroy/Desktop/jetson_camera_imx/vc_mipi_nvidia/build/Nano_32.5.2/Linux_for_Tegra/source/public/kernel/kernel-4.9/scripts/Makefile.build:649: recipe for target 'drivers/media/i2c' failed
make[3]: *** [drivers/media/i2c] Error 2
make[3]: *** Waiting for unfinished jobs....
```

Host hardware: Ubuntu 18.04
Target: Nvidia Jetson Nano 4GB development kit (B01), L4T R32.5.2 
Camera:  IMX327-c 
How can I fix that? I tried with the master and tags: v0.16.0,v0.17.0 and v0.17.1, its the same error in all of them.

Thanks, Roy.

FleetWithMac, 2024-03-25:
Hi try the v0.15.1 tag

bazo80, 2024-04-24:
Hello @RoyAmoyal , hello @FleetWithMac ,

we will insert a small bugfix release, maybe 0.17.2, which will fix that issue. There has been a refactoring and not all of the L4Ts has been tested with it. Sorry for that inconvenience. Basically this is not a major issue, you could simply initialize the tRet variable right after the instantiation:
```
vc_mode tRet;
memset(&tRet, 0, sizeof(tRet));
```