VC-MIPI-modules/vc_mipi_nvidia #89: JN30A 38346-3 sensor_common_parse_image_props:active_l propperty missing / Failed to read mode 0 / Failed to initialise imx219

Frennz, 2024-05-16:
### I Got the JN30 Board with 4 Mipi Lanes 
[image]
Flashing with the Attributes Nano and JN30/JN30D Worked
The DTSI is attached
[tegra210-camera-vc-mipi-cam.zip](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/15341737/tegra210-camera-vc-mipi-cam.zip)

### When powering on this is Displayed both when Mipi Camera is conected to port AB and CD:
[image]

CSI-AB
[image]

CSI-CD

This error message apears; 
sensor_common_parse_image_props:active_l propperty missing
Failed to read mode0
Failed to initialise imx219

### Is my Configuration wrong?
```
/*
 * Copyright (c) 2020-2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>
#include <dt-bindings/platform/t210/t210.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Nano on Auvidea JNX30-LC-PD
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_AB      1   // 1: Enabled, 0: Disabled  (serial_a => CSI Port 0 => VI Stream 0)
#define VC_MIPI_CAM_CD      1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
#define VC_MIPI_CAM_E       0   // 1: Enabled, 0: Disabled  (serial_e => CSI Port 4 => VI Stream 4)
#define VC_MIPI_CAM_F       0   // 1: Enabled, 0: Disabled  (serial_f => CSI Port 5 => VI Stream 5)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX462, IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       4   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
[... 725 lines cut ...]
#endif
            status = "okay";
            reset-gpios = <&gpio CAM4_PWDN GPIO_ACTIVE_HIGH>;
        };
    };
#endif

    gpio@6000d000 {
        camera-trigger-high {
            status = "okay";
            gpio-hog;
            output-high;
            gpios = <CAM_TRIGGER 0>;
            label = "cam-trigger";
        };
 
        camera-control-output-low {
            gpio-hog;
            output-low;
            gpios = < CAM1_PWDN 0  CAM2_PWDN 0>;
            label = "cam1-pwdn", "cam2-pwdn";
        };
    };
};
```

bazo80, 2024-05-17:
Hello @Frennz ,

it seems, that the patches have not been applied correctly. The standard device driver IMX219 is being loaded. It states, that active_l property is missing, so the kernel_common patches have been applied -> that's ok.
The dtsi file seems to be ok, but this dtsi is not being included. In the kernel_Nano_&ast; folder are patches located, which remove the &ast;-imx219.dtsi and adding the tegra210-camera-vc-mipi-cam.dtsi into the whole device tree of the system. This files are called 000&ast;-Modified-tegra210-&ast;.patch
After the system has started, you can call
```
dmesg | grep "DTS File Name"
```
on your running target.
This should give you the used device tree name with the path. Please tell me the dts name.
One of the 000*-Modify*.patch files should modify your *.dts file.
Please lookup your dts in the .../hardware/nvidia/platform/... path and have a look whether there is an #include directive like this:
```
#include "../../porg/kernel-dts/porg-platforms/tegra210-camera-vc-mipi-cam.dtsi"
```
If there isn't a patch file which modifies your *.dts file, then we must setup a patch for exactly this dts-file.
We have tested the whole system with the JNX30D not the JNX30A, because providing all system combinations would mean a huge effort. But I cannot imagine, that the JNX30A is using a different *.dts file.

Btw: Have you used a complete new setup (a new folder with setup/build and flash.sh) for your board?

regards

Frennz, 2024-05-29:
> Hello @Frennz ,
> 
> it seems, that the patches have not been applied correctly. The standard device driver IMX219 is being loaded. It states, that active_l property is missing, so the kernel_common patches have been applied -> that's ok. The dtsi file seems to be ok, but this dtsi is not being included. In the kernel_Nano_* folder are patches located, which remove the *-imx219.dtsi and adding the tegra210-camera-vc-mipi-cam.dtsi into the whole device tree of the system. This files are called 000*-Modified-tegra210-*.patch After the system has started, you can call
> 
> ```
> dmesg | grep "DTS File Name"
> ```
> 
> on your running target. This should give you the used device tree name with the path. Please tell me the dts name. One of the 000*-Modify*.patch files should modify your *.dts file. Please lookup your dts in the .../hardware/nvidia/platform/... path and have a look whether there is an #include directive like this:
> 
> ```
> #include "../../porg/kernel-dts/porg-platforms/tegra210-camera-vc-mipi-cam.dtsi"
> ```
> 
> If there isn't a patch file which modifies your *.dts file, then we must setup a patch for exactly this dts-file. We have tested the whole system with the JNX30D not the JNX30A, because providing all system combinations would mean a huge effort. But I cannot imagine, that the JNX30A is using a different *.dts file.
> 
> Btw: Have you used a complete new setup (a new folder with setup/build and flash.sh) for your board?
> 
> regards

Hello @bazo80, sorry for the late response, I was on vacation.

I Followed your described steps and looked up the DTS Files on my Target Machine with the Following output: 
```
vc@nvidia:~$ dmesg | grep "DTS File Name" 
[    0.212326] DTS File Name: /home/ms/vc_mipi_nvidia/build/Nano_32.7.1/Linux_for_Tegra/source/public/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts
[    0.416281] DTS File Name: /home/ms/vc_mipi_nvidia/build/Nano_32.7.1/Linux_for_Tegra/source/public/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts
```

So I think you want me to take a look at the file tegra210-p3448-0000-p3449-0000-a02.dts in my Build folder on my host machine.?

I flashed my board from a empty folder and a newly cloned repo with all tool chains and everything downloaded again.

This is the Content of the **DTS** file from the BUILD Folder on my HOST machine that was flashed

```
/*
 * arch/arm64/boot/dts/tegra210-p3448-0000-p3449-0000-a02.dts
 *
 * Copyright (c) 2018-2021, NVIDIA CORPORATION.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; version 2 of the License.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License along
 * with this program; if not, write to the Free Software Foundation, Inc.,
 * 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301, USA.
 *
 */

#include "tegra210-porg-p3448-common.dtsi"
#include "porg-platforms/tegra210-porg-camera-rbpcv3-imx477.dtsi"
#include "porg-platforms/tegra210-porg-camera-rbpcv2-imx219.dtsi"
#include "porg-platforms/tegra210-porg-pinmux-p3448-0000-a02.dtsi"
#include "porg-platforms/tegra210-porg-gpio-p3448-0000-a02.dtsi"
#include "porg-platforms/tegra210-porg-p3448-emc-a00.dtsi"

/ {
	model = "NVIDIA Jetson Nano Developer Kit";
	compatible = "nvidia,p3449-0000-a02+p3448-0000-a02", "nvidia,jetson-nano", "nvidia,tegra210";
	nvidia,dtsfilename = __FILE__;

	spi@70410000 {
		status = "okay";
[... 22 lines cut ...]
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "GPIO01",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "GPIO07",   "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "GPIO12",       "",     "",     "",     "",     "",
		"GPIO11",      "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "GPIO09",     "",     "",     "",
		"",     "",     "",     "",     "",     "",     "",     "",     "",     "",
		"",     "",     "SPI1_CS1",     "",     "",     "",     "",     "",     "",     "";
	};
};
```
Also, A pended Here [tegra210-p3448-0000-p3449-0000-a02.zip](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/files/15477479/tegra210-p3448-0000-p3449-0000-a02.zip)

So that is that. I Hoped I Followed your steps correctly, and you can help me to configure the device tree for my system and figure my problem out.

thanks for the help!

If I misunderstood something, I apologize, followup instructions could also be in German if it's also your mother language. 

with the best regards

bazo80, 2024-05-29:
Hello @Frennz ,

if you have a look into the folder build/Nano_XX.Y.Z/Linux_for_Tegra/source/public/hardware/nvidia/platform/t210/porg/kernel-dts/
you will find several dts files. Two of them are interesting for your use case:
tegra210-p3448-0000-p3449-0000-<b>a02.dts</b> and tegra210-p3448-0000-p3449-0000-<b>b00.dts</b>
The a02.dts file will be used for your board (JN30A), the b00.dts file is used for my setup (Nano on JNX30). 

When you compare both files, you will see that the b00.dts has a compiler switch included, which removes both standard sensors (IMX219 and IMX477) from the device tree file and includes the vc_mipi dtsi (tegra210-camera-vc-mipi-cam.dtsi) like this:
<pre>
...
#define VC_MIPI 1

#include "tegra210-porg-p3448-common.dtsi"
#if VC_MIPI
	#include "porg-platforms/tegra210-camera-vc-mipi-cam.dtsi"
#else
	#include "porg-platforms/tegra210-porg-camera-rbpcv3-<b>dual</b>-imx477.dtsi"
	#include "porg-platforms/tegra210-porg-camera-rbpcv2-<b>dual</b>-imx219.dtsi"
#endif
#include "porg-platforms/tegra210-porg-pinmux-p3448-0000-<b>b00</b>.dtsi"
#include "porg-platforms/tegra210-porg-gpio-p3448-0000-<b>b00</b>.dtsi"
#include "porg-platforms/tegra210-porg-p3448-emc-a00.dtsi"

/ {
	model = "NVIDIA Jetson Nano Developer Kit";
	compatible = "nvidia,p3449-0000-<b>b00</b>+p3448-0000-<b>b00</b>", "nvidia,jetson-nano", "nvidia,tegra210";
	nvidia,dtsfilename = __FILE__;
...
</pre>

Your a02.dts file is not patched, so you have to insert the vc_mipi dtsi.
When you apply this mechanism to your file, then it should look like this:
<pre>
...
#define VC_MIPI 1

#include "tegra210-porg-p3448-common.dtsi"

#if VC_MIPI
	#include "porg-platforms/tegra210-camera-vc-mipi-cam.dtsi"
#else
        #include "porg-platforms/tegra210-porg-camera-rbpcv3-imx477.dtsi"
        #include "porg-platforms/tegra210-porg-camera-rbpcv2-imx219.dtsi"
#endif
#include "porg-platforms/tegra210-porg-pinmux-p3448-0000-<b>a02</b>.dtsi"
#include "porg-platforms/tegra210-porg-gpio-p3448-0000-<b>a02</b>.dtsi"
#include "porg-platforms/tegra210-porg-p3448-emc-a00.dtsi"

/ {
	model = "NVIDIA Jetson Nano Developer Kit";
	compatible = "nvidia,p3449-0000-<b>a02</b>+p3448-0000-<b>a02</b>", "nvidia,jetson-nano", "nvidia,tegra210";
	nvidia,dtsfilename = __FILE__;
...
</pre>

Please be careful, every number/letter, especially in the filenames is important!
This process needs to be done one time. After that, you can modify your tegra210-camera-vc-mipi-cam.dtsi with ...
```
./setup.sh -c
```
... compile the device tree with ...
```
./build.sh -d
```
...and flash it with...
```
./flash.sh -d
```
After the system boot up, the mipi driver should be loaded.

---

Also prinzipielles Vorgehen ist, dass das setup Skript unsere Patches auf den Source-Code und auf die von NVIDIA bereits mitgelieferten Device-Tree-Dateien anwendet. Je nach Modul-Board Kombination wird das entsprechende vc-mipi-cam.dtsi (Device-Tree-Include-File) in den entsprechenden Device-Tree eingebunden. 
In Ihrem Fall wäre das die Datei tegra210-camera-vc-mipi-cam.dtsi aus dem Auvidea_JNX30_Nano Verzeichnis in den Device-Tree von dem JN30A tegra210-p3448-0000-p3449-0000-<b>a02.dts</b>. Wenn Sie sich diese Datei im Verzeichnis build/Nano_XX.Y.Z/Linux_for_Tegra/source/public/hardware/nvidia/platform/t210/porg/kernel-dts/ anschauen, stehen darin zwei Kamera-Include Files (IMX219 und IMX477). Das sind Standard-Sensoren, respektive Beispiele. Diese müssen mittels Schalter rausgenommen werden und durch die Include-Direktive tegra210-camera-vc-mipi-cam.dtsi ersetzt werden. Ihre anzupassende Device-Tree-Datei (tegra210-p3448-0000-p3449-0000-<b>a02.dts</b>) muss dann wie folgt aussehen:
<pre>
...
#define VC_MIPI 1

#include "tegra210-porg-p3448-common.dtsi"

#if VC_MIPI
	#include "porg-platforms/tegra210-camera-vc-mipi-cam.dtsi"
#else
        #include "porg-platforms/tegra210-porg-camera-rbpcv3-imx477.dtsi"
        #include "porg-platforms/tegra210-porg-camera-rbpcv2-imx219.dtsi"
#endif
#include "porg-platforms/tegra210-porg-pinmux-p3448-0000-<b>a02</b>.dtsi"
#include "porg-platforms/tegra210-porg-gpio-p3448-0000-<b>a02</b>.dtsi"
#include "porg-platforms/tegra210-porg-p3448-emc-a00.dtsi"

/ {
	model = "NVIDIA Jetson Nano Developer Kit";
	compatible = "nvidia,p3449-0000-<b>a02</b>+p3448-0000-<b>a02</b>", "nvidia,jetson-nano", "nvidia,tegra210";
	nvidia,dtsfilename = __FILE__;
...
</pre>

Das muss auch nur einmalig gemacht werden. Im Anschluss kann die tegra210-camera-vc-mipi-cam.dtsi Datei dann mittels `./setup.sh -c` angepasst werden.
Der `./build.sh -d` Befehl kopiert dann die tegra210-camera-vc-mipi-cam.dtsi Datei in das porg-platforms Verzeichnis, von wo aus sie eingebunden werden kann und der komplette Device-Tree wird gebaut.
Zur Kontrolle kann man den kompletten Device-Tree auch wieder zurückwandeln. Im Verzeichnis build/Nano_XX.Y.Z/Linux_for_Tegra/kernel/dtb wäre das in dem Fall die Datei tegra210-p3448-0000-p3449-0000-a02.dtb
Mit dem folgendem Befehl erhält man den zurückgewandelten Device-Tree:
`dtc -I dtb -O dts tegra210-p3448-0000-p3449-0000-a02.dtb -o tegra210-p3448-0000-p3449-0000-a02.dts`
In der darauf entstandenen tegra210-p3448-0000-p3449-0000-a02.dts Datei sollte weder IMX219 noch IMX477 vorkommen, sondern nur der Eintrag vc_mipi. Wenn das der Fall ist, können Sie mit `./flash.sh -d` den Device-Tree flashen.

Ich kann es leider nicht selber testen, da ich das JN30A nicht zur Hand habe

Frennz, 2024-06-04:
Hi @bazo80 I followed your instructions and flashed the board with the modifications you instructed.
I think they applied correctly, but other things seem not to bind correctly, if you can give me insight how to patch these I would be very grateful! 
[image]
These Errors arose after modifying the tegra210-p3448-0000-p3449-0000-a02.dts file.
(I followed your instructions and also used a fresh clone with build --all with the modifications same result) 

On a different Note
The difference between the JN30 (JN30B probably because the patches only apply to b01 dts files) and my board is that the JN30A only support old A revision compute modules and not the newer B01 Compute modules from the B01 Development board that you support. I sadly cannot use the B01 Module on the 4- Lane Board because of the difference of the pin outs between these two revisions.
[image]
I hope this does not play a mayor role and is unrelated to the Boot errors I'm getting, and you can still help me out to get it working, and I hope that this isn't a big factor that would include too much work to get it working.
I acknowledge that it's a different chipset with a different pin structure, and I would understand if support for that platform from your side is limited.

Regards

Frennz, 2024-06-06:
@bazo80 is The JNX30D a board that is confirmed and tested with the B01 module? that is working with the IMx412?

bazo80, 2024-06-07:
Hello @Frennz ,

colleague of mine could make a statement about the JNX30<b>D</b>, but he is in vacation until monday.
I can confirm the Nano B01 module (Model: P3448 180-13448-DAAA-B01) running on 
JNX30 - PSE SKU: 70803 (2 cams) and 
JNX42-LM SKU: 70784-LM (1 cam). 
The JNX42 is designed for the Orins, but you can also attach the Nano B01 with one single cam (cam0).
https://auvidea.eu/download/manual/JNX42/JNX42_Manual.pdf 
page 6:
"(Nano, TX2NX and Xavier NX will only support one USB 3.0 port and only one CSI-2 interface.)"

I've just tested the IMX412 with Nano B01 on JNX30 and JNX42 => 4 lanes, 40Hz, working.

------
BTW: your last dmesg stated, that the system tried to load two instances of the vc_mipi driver
```
vc_mipi 0-001a: vc_mod_setup(): Unable to get...
vc_mipi 6-001a: vc_mod_setup(): Unable to get...
```
That means, that the patch for the driver took effect.
On the firmware site 
https://auvidea.eu/firmware/
I have seen a note which states:
"JN30A: only 1x CSI-2 camera (imx219) is supported at this time."

You could try to define only one sensor in your dtsi, maybe the system behaves in another way. Hard to say, without having this old setup.

regards

Frennz, 2024-06-09:
@bazo80 
Sadly It's still not working even with:
`#define VC_MIPI_CAM_AB 1 // 1: Enabled, 0: Disabled (serial_a => CSI Port 0 => VI Stream 0)
#define VC_MIPI_CAM_CD 0 // 1: Enabled, 0: Disabled (serial_c => CSI Port 2 => VI Stream 2)
#define VC_MIPI_CAM_E 0 // 1: Enabled, 0: Disabled (serial_e => CSI Port 4 => VI Stream 4)
#define VC_MIPI_CAM_F 0 // 1: Enabled, 0: Disabled (serial_f => CSI Port 5 => VI Stream 5)`
as you see here
same output than before
[image]

If you can confirm the JNX30D with your colleague, that would be helpful, else I'll try to backorder the JNX30 I guess.
Thanks for help in any case.

pmliquify, 2024-06-11:
Hello @Frennz,

I repeated my test to confirm to you that the following setup works fine:
- Carrier board: Auvidea JNX30D
- SoM: NVIDIA Jetson Nano B01
- Camera: Vision Components IMX412
- Driver: VC MIPI NVIDIA Version 0.17.1
- Camera port: 1 + 2 with 2 and 4 lanes

[image]

Regards,
Peter

Frennz, 2024-06-11:
> Hello @Frennz,
>
>
>
> I repeated my test to confirm to you that the following setup works fine:
>
> - Carrier board: Auvidea JNX30D
>
> - SoM: NVIDIA Jetson Nano B01
>
> - Camera: Vision Components IMX412
>
> - [...]

Thanks I will purchase this then. Thank you and bazo80 for the amazing support!

closed 2024-06-11 by Frennz