VC-MIPI-modules/vc_mipi_nvidia #96: VC Driver reports Error -34 on boot

s-withers-archangel, 2024-06-06:
After running the quickstart script and booting the JETSON i'm not getting any video devices under `/dev` and the VC driver is reporting an error in the dmesg output
```
[    7.555172] vc_mipi 9-001a: tegracam sensor driver:vc_mipi_v2.0.6
[    7.565140] vc_mipi 9-001a: ctrl Frame Rate range update failed
[    7.575227] vc_mipi 9-001a: Error -34 updating mode specific control ranges 
[    7.582472] vc_mipi 9-001a: Error -34 updating control ranges
[    7.588374] vc_mipi 9-001a: Failed to init ctrls vc_mipi
```

Hardware Setup:
```
Host: Ubuntu 22.04.3 LTS
JETSON: NVIDIA Jetson Orin Nano 8GB NVME (devkit)
Camera:  IMX568C
Jetpack: 35.3.1
```
Steps followed
```
git clone https://github.com/VC-MIPI-modules/vc_mipi_nvidia.git
cd vc_mipi_nvidia/bin
./quickstart.sh
Selected jetson board and jetpack version
```
Modified `tegra234-camera-vc-mipi-cam.dtsi` is follows:
```
/*
 * Copyright (c) 2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano on NVIDIA Jetson Orin Nano Developer Kit
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_b => CSI Port 1 => VI Stream 0)
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX462, IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       2   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "4" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
[... 532 lines cut ...]
[    7.353891] i2c 9-0010: |  9 |    1188 |       4 | RAW08   | EXT.TRG |       0 |
[    7.353895] i2c 9-0010: | 10 |    1188 |       4 | RAW10   | EXT.TRG |       0 |
[    7.365983] i2c 9-0010: | 11 |    1188 |       4 | RAW12   | EXT.TRG |       0 |
[    7.365986] i2c 9-0010: +----+---------+---------+---------+---------+---------+
[    7.381252] i2c 9-0010: vc_init_ctrl_imx568(): Initialising module control for IMX568
[    7.393074] i2c 9-0010: +-------+--------+------------+-----------+
[    7.393080] i2c 9-0010: | lanes | format | exposure   | framerate |
[    7.429588] i2c 9-0010: |       |        | max [us]   | max [mHz] |
[    7.436016] i2c 9-0010: +-------+--------+------------+-----------+
[    7.442445] i2c 9-0010: |     2 | RAW08  |  148226464 |     51778 |
[    7.448875] i2c 9-0010: |     2 | RAW10  |  183022398 |     41933 |
[    7.455313] i2c 9-0010: |     2 | RAW12  |  218036426 |     35200 |
[    7.461751] i2c 9-0010: |     4 | RAW08  |   78617813 |     97627 |
[    7.461753] i2c 9-0010: |     4 | RAW10  |   96015783 |     79936 |
[    7.461754] i2c 9-0010: |     4 | RAW12  |  113413743 |     67672 |
[    7.461755] i2c 9-0010: +-------+--------+------------+-----------+
[    7.461760] i2c 9-0010: VC MIPI Core successfully initialized
[    7.507782] i2c 9-0010: vc_mod_set_trigger_mode(): Set trigger mode: DISABLED
[    7.523520] i2c 9-0010: vc_mod_set_io_mode(): Set IO mode: DISABLED
[    7.601391] i2c i2c-2: Added multiplexed i2c bus 9
[   10.028245] i2c i2c-2: Added multiplexed i2c bus 10
[   10.033258] i2c-mux-gpio cam_i2cmux: 2 port mux on 3180000.i2c adapter
```

Any help much appreciated

bazo80, 2024-06-10:
Hello @s-withers-archangel ,

you have set:
<pre>
max_framerate            = "41300";     // mHz
...
default_framerate        = "60000";     // mHz
</pre>

The default_framerate must be less or equal than the max_frame_rate. You should set the 
```
default_framerate        = "41300";     // mHz
```
for both sensors.

Please have a try.

best regards

s-withers-archangel, 2024-06-11:
That was the cause, thank you!

closed 2024-06-11 by s-withers-archangel