VC-MIPI-modules/vc_mipi_nvidia #103: L4T 36 issues with IMX568 and IMX566

kkhakixyz, 2024-06-20:
Hi,

We are using Jetson L47 36.3, and therefore have been using the branch 77-support-nvidia-l4t-362, and are on commit 375a4be. We are seeing some interesting behaviour, and there seems to be a bug in the driver.

Please find the technical steps below:

- When the system boots up, the camera is detected and initialised. And the camera view appears and seems to work well (colors being a bit washed out) when we run the command `gst-launch-1.0 nvarguscamerasrc sensor_id=0 ! 'video/x-raw(memory:NVMM),width=2464, height=2064, framerate=30/1, format=NV12' ! nvvidconv flip-method=0 ! queue ! autovideosink`. The terminal shows:
```shell
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 2464 x 2064 FR = 41.299998 fps Duration = 24213076 ; Analog Gain range min 0.000000, max 48.000000; Exposure Range min 1000, max 4000000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 0 
   Output Stream W = 2464 H = 2064 
   seconds to Run    = 0 
   Frame Rate = 41.299998 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
Redistribute latency...
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:09.513306338
Setting pipeline to NULL ...
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
Freeing pipeline ...
``` 

- However, if `journalctl -f` is run on another terminal, it seems there is an issue. The messages below are repeated continously whilst the camera is streaming:
```shell
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 2 : FuSaCaptureViErrorDecode VI Error in frame:  0
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 2 : FALCON_ERROR 0x00000e
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 5 : ChanselFault 0x040000
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 5 : Current line in frame
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 5 : [hi:lo] :  31 16
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 5 : Decoder Field Value:  4
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 5 :
Jun 20 12:02:40 xyzreality-orin4 nvargus-daemon[2933]: Module_id 30 Severity 2 : EMB_SHORT:
``` 

- We then set it up to debug further using the following
```shell
echo 1 > /sys/kernel/debug/tracing/tracing_on
echo 30720 > /sys/kernel/debug/tracing/buffer_size_kb
echo 1 > /sys/kernel/debug/tracing/events/tegra_rtcpu/enable
echo 1 > /sys/kernel/debug/tracing/events/freertos/enable
echo 3 > /sys/kernel/debug/camrtc/log-level
echo 1 > /sys/kernel/debug/tracing/events/camera_common/enable
echo > /sys/kernel/debug/tracing/trace
sudo modprobe rtcpu_debug
``` 
- And then if the camera is streamed and then stopped, the trace file seems to have no issues 
[trace log 1.log](https://github.com/user-attachments/files/15916889/trace.log.1.log)

- If we enable further logging and run nvargus through the terminal using the following
```shell
sudo su
export enableCamPclLogs=5
export enableCamScfLogs=5
/usr/sbin/nvargus-daemon
``` 

- This time when the gstreamer command is run, a black screen is displayed for a few seconds, and then it closes. The terminal shows:
```shell
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 2464 x 2064 FR = 41.299998 fps Duration = 24213076 ; Analog Gain range min 0.000000, max 48.000000; Exposure Range min 1000, max 4000000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 0 
   Output Stream W = 2464 H = 2064 
   seconds to Run    = 0 
   Frame Rate = 41.299998 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 277)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 350)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 366)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 379)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadExecute:732 NvBufSurfaceFromFd Failed.
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:243 (propagating)
Redistribute latency...
Got EOS from element "pipeline0".
Execution ended after 0:00:01.014790620
Setting pipeline to NULL ...
GST_ARGUS: Cleaning up
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
(Argus) Error InvalidState: Receive thread is not running cannot send. (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 96)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
Caught SIGSEGV
Segmentation fault (core dumped)
``` 

- the nvargus log is attached 
[nvargus log.txt](https://github.com/user-attachments/files/15916943/nvargus.log.txt)

- `journalctl -f` still has the same messages, although only a few since the stream crashed.

- the trace file this time reports quite a few issues 
[trace log 2.log](https://github.com/user-attachments/files/15916920/trace.log.2.log)

kkhakixyz, 2024-06-21:
This has now been resolved.

The modifications required to fix it for L4T 36.3 are:

In [vc_mipi_modules.c](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/blob/77-support-nvidia-l4t-362/src/driver/vc_mipi_modules.c) in `vc_init_ctrl_imx568`, replace line 816 with `FRAME(0, 0, 2472, 2064)`.

In [tegra234-p3767-camera-p3768-vc_mipi-dual.dts](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/blob/77-support-nvidia-l4t-362/src/devicetree/NV_DevKit_OrinNano/tegra234-p3767-camera-p3768-vc_mipi-dual.dts) `VC_MIPI_METADATA_H` needs to be `"1"` for IMX566.

bazo80, 2024-06-24:
Hello @kkhakixyz ,

thank you for your investigation. We'll have a look into that width / height / metadata height topic in our next sprint iteration.

best regards