VC-MIPI-modules/vc_mipi_nvidia #112: Configuration VC_MIPI_CAM_0=0 and VC_MIPI_CAM_1=1

fre-wolf, 2024-07-24:
Hello,
We have an application where it is easier to connect a single camera to our mipi port 1 instead of 0. Is that configuration supported? If I use the defines accordingly in the dtsi file, I get a camera that is detected correctly in dmesg, but no ``video0`` or ``video1`` device in ``/dev/``. Using a dtsi configuration where both (identical) cameras are active works, but throws warnings : ``[ 5.547274] tegra-vii2c 546c0000.i2c: no acknowledge from address 0x10``, which is expected as that camera isn't plugged in.
Thank you for your help.

OctaAIVision, 2024-08-13:
hi 
i have kinda similar issue did u solve it?

fre-wolf, 2024-08-16:
Hi,
at the moment, I just accept the warning as it is... One workaround is possibly to reassign the MIPI lanes in dtsi files, but I don't think it's worth it...