VC-MIPI-modules/vc_mipi_nvidia #121: Second camera sensor not working on Jetpack 6 (two IMX412)

fredriktegnell, 2024-11-07:
L4T version: 36.2.0 DP

When having two IMX412 connected to the Jetson Orin Nano devkit the video0 one works well but the video1 seems to time out, here is the bootlog messages:
[ 3174.706552] vc_mipi 9-001a: vc_core_set_format(): Set format 0x300f (RG10)
[ 3174.706565] vc_mipi 9-001a: vc_core_try_format(): Try format 0x300f (RG10, format: 0x2b)
[ 3174.706710] vc_mipi 9-001a: vc_update_image_size_from_mode(): Update image size from mode0 (l: 0, t: 0, w: 4032, h: 3040)
[ 3174.706727] i2c 9-0010: vc_mod_set_mode(): Set module mode: 1 (lanes: 4, format: RG10, type: STREAM)
[ 3174.706732] i2c 9-0010: vc_mod_set_power(): Set module power: down
[ 3174.707079] i2c 9-0010: vc_mod_set_power(): Set module power: up
[ 3175.313129] vc_mipi 9-001a: vc_sen_set_exposure(): Set sensor exposure: 10000 us
[ 3175.313831] vc_mipi 9-001a: vc_sen_set_gain(): Set sensor gain: 0
[ 3175.314110] vc_mipi 9-001a: vc_core_get_num_lanes(): Get number of lanes: 4
[ 3175.314113] vc_mipi 9-001a: vc_sen_set_blacklevel(): Set sensor black level: 3906 (40/1023)
[ 3175.314386] vc_mipi 9-001a: vc_sen_start_stream(): Start streaming
[ 3176.947439] vc_mipi 9-001a: vc_sen_stop_stream(): Stop streaming
[ 3179.001215] vc_mipi 10-001a: vc_core_set_format(): Set format 0x300f (RG10)
[ 3179.001227] vc_mipi 10-001a: vc_core_try_format(): Try format 0x300f (RG10, format: 0x2b)
[ 3179.001241] vc_mipi 10-001a: vc_update_image_size_from_mode(): Update image size from mode0 (l: 0, t: 0, w: 4032, h: 3040)
[ 3179.001257] i2c 10-0010: vc_mod_set_mode(): Set module mode: 1 (lanes: 4, format: RG10, type: STREAM)
[ 3179.001262] i2c 10-0010: vc_mod_set_power(): Set module power: down
[ 3179.001821] i2c 10-0010: vc_mod_set_power(): Set module power: up
[ 3179.610287] vc_mipi 10-001a: vc_sen_set_exposure(): Set sensor exposure: 10000 us
[ 3179.610872] vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 0
[ 3179.611169] vc_mipi 10-001a: vc_core_get_num_lanes(): Get number of lanes: 4
[ 3179.611174] vc_mipi 10-001a: vc_sen_set_blacklevel(): Set sensor black level: 3906 (40/1023)
[ 3179.611470] vc_mipi 10-001a: vc_sen_start_stream(): Start streaming
[ 3179.611473] vc_mipi 10-001a: vc_sen_stop_stream(): Stop streaming
[ 3184.019972] tegra-camrtc-capture-vi tegra-capture-vi: uncorr_err: request timed out after 5000 ms
[ 3184.019997] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: attempting to reset the capture channel
[ 3184.021466] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: successfully reset the capture channel
[ 3189.139849] tegra-camrtc-capture-vi tegra-capture-vi: uncorr_err: request timed out after 5000 ms
[ 3189.139872] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: attempting to reset the capture channel
[ 3189.140786] tegra-camrtc-capture-vi tegra-capture-vi: err_rec: successfully reset the capture channel
[ 3189.141033] vc_mipi 10-001a: vc_sen_stop_stream(): Stop streaming

We need two cameras to work for our product so please give me a fix asap.

bazo80, 2024-11-07:
Hello @fredriktegnell,

it seems that you have configured both cams with 4 lanes, the nvidia orin nano devkit supports 2 lanes@cam0 and upto 4 lanes@cam1. Either you set the VC_MIPI_LANES define to 2 in order to get both cams working with the same parameters (don't forget about the max_framerate, default_framerate) or you differentiate both sections according to the VC_MIPI_LANE define and the values, which are depending on this define. I guess, we should document that board a little bit better.

regards

fredriktegnell, 2024-11-07:
Hello @bazo80 

I missed that completely, we are now currently rebuilding the image using 2 lanes for each cam to try it out, thank you for the quick response!