VC-MIPI-modules/vc_mipi_nvidia #122: When starting the imx265c camera on Orin NX, the image is pink instead of color

voyager556321, 2024-11-15:
When I start the camera on orin NX when the stream is displayed, it has a purple color instead of different colors, how can this be fixed, and what exactly is the problem, I changed wbmode, but the stream only changed from purple to white because the daylylight parameter, I made the appropriate changes to the device tree for the camera
[image]

bazo80, 2025-02-19:
Hello @voyager556321 ,

there is the so called black_level v4l2 control and the ISP is not aware of that control. Since the driver is initially setting the appropriate blacklevel value for the attached sensor (if the sensor supports that), every pixel has a certain offset. So the ISP tries to compensate this offset.

```
v4l2-ctl -d /dev/videoX -c black_level=1
v4l2-ctl -d /dev/videoX -c black_level=0
```
The other cause might be a wrong bayer pattern, which must be set in the device tree. RGGB/GBRG

Which sensor are you using?

best regards

voyager556321, 2025-02-19:
Thanks, i fixed it a long time ago

closed 2025-02-19 by voyager556321