VC-MIPI-modules/vc_mipi_nvidia #125: Support for L4T 36.4

Qbicz, 2024-12-13:
Hi! The latest release of Nvidia BSP is 36.4 Jetson Linux.

I've seen https://github.com/VC-MIPI-modules/vc_mipi_nvidia/issues/97 but it will be good to target 36.4 since it has some important security features.

I'm especially interested in Orin NX + IMX296, using BSP 36.4.

Qbicz, 2025-01-17:
Hi! Bumping up the question, and even if you haven't tested with 36.4, do you expect a lot of changes between 36.2 and 36.4?

bazo80, 2025-02-11:
Hello @Qbicz ,

sorry for the late response.
There is a side branch https://github.com/VC-MIPI-modules/vc_mipi_nvidia/tree/bazo_364_pre which is intended for L4T 36.4

best regards

Qbicz, 2025-02-11:
Thank you @bazo80 , we're reviewing it and will come back if we encounter any issues.