VC-MIPI-modules/vc_mipi_nvidia #126: Quickstart script fails - not in a git directory

mstanisz0, 2024-12-18:
Hi all,

I'm trying to run the quickstart script with the following settings:

* IMX900
* NVIDIA Jetson Orin Nano 8GB SD (devkit)
* NVIDIA Jetson Orin Nano Developer Kit
* NVIDIA L4T 35.4.1
* VC_MIPI_CAM_1 enabled only
* VC_MIPI_LANES 4
* VC_MIPI_METADATA_H "1"

After about 5 minutes, the script fails with the following error:

```
...
hardware/nvidia/soc/tegra/kernel-include/dt-bindings/clock/r8a7791-clock.h
hardware/nvidia/soc/tegra/kernel-include/dt-bindings/clock/lsi,axm5516-clks.h
hardware/nvidia/soc/tegra/kernel-include/dt-bindings/clock/imx6ul-clock.h
hardware/nvidia/soc/tegra/kernel-include/dt-bindings/clock/exynos4415.h
nvcommon_build.sh
nvbuild.sh
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint:
hint:   git config --global init.defaultBranch <name>
hint:
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint:
hint:   git branch -m <name>
Initialized empty Git repository in /home/matt/imx900-testing/vc_mipi_nvidia/build/Xavier_35.4.1/Linux_for_Tegra/source/public/.git/
fatal: not in a git directory
```

Any thoughts?

Thanks!
Matt

bazo80, 2024-12-19:
Hello @mstanisz0 ,

the quickstart.sh script is calling setup.sh --host. You can also try to run:
```
./setup.sh --host
```
as a standalone command.
The setup.sh and the quickstart.sh script must not be run as root!

Which git version are you using and what is your Ubuntu version?

best regards

mstanisz0, 2024-12-19:
Thanks @bazo80, I found by running as non-root (as you mentioned) and also bumping to the earlier 35.x release helped the script to complete.  I'm on Ubuntu 22.04 (tried both WSL and a PC) and Git version is 2.34.1.  I'm guessing it's the root access that caused the issue.

I'm still seeing an issue with flashing where my board doesn't detect after imaging, but I've run into this before with the stock Nvidia image, so I'll keep debugging.  I think I'm all set for now, thanks!

closed 2024-12-19 by mstanisz0