VC-MIPI-modules/vc_mipi_nvidia #128: IMX900 Gstreamer to Display Not Working

mstanisz0, 2025-01-02:
Hi,

I'm trying to run the IMX900 module in RAW10 mode with 4 MIPI lanes.  I've flashed my Orin Nano DK following the quick start instructions, including adding the modifications to the dt to support gstreamer.

I'm detecting the camera on the CAM1 port:

```
[    4.377333] i2c /dev entries driver
[    4.381056] tegra-i2c 3160000.i2c: Adding to iommu group 2
[    4.386887] tegra-i2c c240000.i2c: Adding to iommu group 2
[    4.392571] tegra-i2c 3180000.i2c: Adding to iommu group 2
[    4.398251] tegra-i2c 3190000.i2c: Adding to iommu group 2
[    4.403927] tegra-i2c 31b0000.i2c: Adding to iommu group 2
[    4.409623] tegra-i2c 31c0000.i2c: Adding to iommu group 2
[    4.415302] tegra-i2c c250000.i2c: Adding to iommu group 2
[    4.420984] tegra-i2c 31e0000.i2c: Adding to iommu group 2
[    9.763391] i2c i2c-2: Added multiplexed i2c bus 9
[    9.875289] i2c 10-0010: +--- VC MIPI Camera -----------------------------------+
[    9.882990] i2c 10-0010: | MANUF. | Vision Components               MID: 0x0427 |
[    9.890680] i2c 10-0010: | MODULE | ID:  0x0900                     REV:   0002 |
[    9.898375] i2c 10-0010: | SENSOR | SONY IMX900C                                |
[    9.906076] i2c 10-0010: +--------+---------------------------------------------+
[    9.913776] i2c 10-0010: +--- Sensor Registers ------+--------+--------+--------+
[    9.921474] i2c 10-0010: |                           | low    | mid    | high   |
[    9.929163] i2c 10-0010: +---------------------------+--------+--------+--------+
[    9.936857] i2c 10-0010: | idle                      | 0x3000 |        |        |
[    9.944557] i2c 10-0010: | horizontal start          | 0x3120 | 0x3121 |        |
[    9.952247] i2c 10-0010: | vertical start            | 0x3122 | 0x3123 |        |
[    9.959944] i2c 10-0010: | horizontal end            | 0x0000 | 0x0000 |        |
[    9.967636] i2c 10-0010: | vertical end              | 0x30d0 | 0x30d1 |        |
[    9.975333] i2c 10-0010: | hor. output width         | 0x3124 | 0x3125 |        |
[    9.983020] i2c 10-0010: | ver. output height        | 0x3126 | 0x3127 |        |
[    9.990723] i2c 10-0010: | exposure                  | 0x3240 | 0x3241 | 0x3242 |
[    9.998415] i2c 10-0010: | gain                      | 0x3514 | 0x3515 |        |
[   10.006121] i2c 10-0010: +---------------------------+--------+--------+--------+
[   10.013814] i2c 10-0010: | clock for ext. trigger    | 54000000 Hz              |
[   10.021507] i2c 10-0010: | pixel clock               | 74250000 Hz              |
[   10.029201] i2c 10-0010: | shutter offset            |      115 us              |
[   10.036896] i2c 10-0010: +---------------------------+--------------------------+
[   10.044584] i2c 10-0010: +--- Module Modes -------+---------+---------+---------+
[   10.052284] i2c 10-0010: |  # | rate    | lanes   | format  | type    | binning |
[   10.059979] i2c 10-0010: +----+---------+---------+---------+---------+---------+
[   10.067681] i2c 10-0010: |  0 |    1485 |       2 | RAW08   | STREAM  |       0 |
[   10.075376] i2c 10-0010: |  1 |    1485 |       2 | RAW10   | STREAM  |       0 |
[   10.083071] i2c 10-0010: |  2 |    1485 |       2 | RAW12   | STREAM  |       0 |
[   10.090769] i2c 10-0010: |  3 |    1485 |       2 | RAW08   | EXT.TRG |       0 |
[   10.098466] i2c 10-0010: |  4 |    1485 |       2 | RAW10   | EXT.TRG |       0 |
[   10.106164] i2c 10-0010: |  5 |    1485 |       2 | RAW12   | EXT.TRG |       0 |
[   10.113857] i2c 10-0010: |  6 |    1485 |       4 | RAW08   | STREAM  |       0 |
[   10.121549] i2c 10-0010: |  7 |    1485 |       4 | RAW10   | STREAM  |       0 |
[   10.129243] i2c 10-0010: |  8 |    1485 |       4 | RAW12   | STREAM  |       0 |
[   10.136942] i2c 10-0010: |  9 |    1485 |       4 | RAW08   | EXT.TRG |       0 |
[   10.144629] i2c 10-0010: | 10 |    1485 |       4 | RAW10   | EXT.TRG |       0 |
[   10.152324] i2c 10-0010: | 11 |    1485 |       4 | RAW12   | EXT.TRG |       0 |
[   10.160022] i2c 10-0010: +----+---------+---------+---------+---------+---------+
[   10.167719] i2c 10-0010: vc_init_ctrl_imx900(): Initialising module control for IMX900
[   10.175881] i2c 10-0010: +-------+--------+------------+-----------+
[   10.182422] i2c 10-0010: | lanes | format | exposure   | framerate |
[   10.188953] i2c 10-0010: |       |        | max [us]   | max [mHz] |
[   10.195484] i2c 10-0010: +-------+--------+------------+-----------+
[   10.202017] i2c 10-0010: |     2 | RAW08  |  103934233 |     88888 |
[   10.208557] i2c 10-0010: |     2 | RAW10  |  127422196 |     72505 |
[   10.215085] i2c 10-0010: |     2 | RAW12  |  150708833 |     61300 |
[   10.221627] i2c 10-0010: |     4 | RAW08  |   76369432 |    120977 |
[   10.228158] i2c 10-0010: |     4 | RAW10  |   82241422 |    112334 |
[   10.234698] i2c 10-0010: |     4 | RAW12  |  137824007 |     67033 |
[   10.241225] i2c 10-0010: +-------+--------+------------+-----------+
[   10.247757] i2c 10-0010: VC MIPI Core successfully initialized
[   10.260467] i2c 10-0010: vc_mod_set_trigger_mode(): Set trigger mode: DISABLED
[   10.267897] i2c 10-0010: vc_mod_set_io_mode(): Set IO mode: DISABLED
[... 54 lines cut ...]
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:751 No cameras available
/GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)2048, height=(int)1536, format=(string)NV12, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)2048, height=(int)1536, format=(string)NV12, framerate=(fraction)30/1
/GstPipeline:pipeline0/Gstnvvconv:nvvconv0.GstPad:src: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:src: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstXvImageSink:xvimagesink0.GstPad:sink: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstTextOverlay:fps-display-text-overlay.GstPad:video_sink: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0.GstGhostPad:sink: caps = video/x-raw, width=(int)2048, height=(int)1536, framerate=(fraction)30/1, format=(string)YUY2
/GstPipeline:pipeline0/Gstnvvconv:nvvconv0.GstPad:sink: caps = video/x-raw(memory:NVMM), width=(int)2048, height=(int)1536, format=(string)NV12, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw(memory:NVMM), width=(int)2048, height=(int)1536, format=(string)NV12, framerate=(fraction)30/1
/GstPipeline:pipeline0/GstFPSDisplaySink:fpsdisplaysink0/GstXvImageSink:xvimagesink0: sync = false
Got EOS from element "pipeline0".
Execution ended after 0:00:00.010421477
Setting pipeline to NULL ...
Freeing pipeline ...
```

It seems like the `751 No cameras available` might be the error, so I'm assuming I have my DT configured incorrectly.

What is causing this issue?  I've tried forcing Ubuntu to boot into CLI so there's nothing on my display.

Matt

mstanisz0, 2025-01-02:
Here's my .dtsi file from my host system:

```
/*
 * Copyright (c) 2023, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano on NVIDIA Jetson Orin Nano Developer Kit
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_b => CSI Port 1 => VI Stream 0)
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX462, IMX565, IMX566, IMX567, IMX568, IMX900
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       4   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462, IMX900
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "1" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
//  this repository https://github.com/VC-MIPI-modules/vc_mipi_nvidia#gstreamer-support
// ------------------------------------------------------------------------------------------------

#if VC_MIPI_LANES == 1
        #define BUS_WIDTH 1
        #define NUM_LANES "1" 
        #define NUM_CSI_LANES 2
        #define PIX_CLK_HZ "150000000"
#endif
#if VC_MIPI_LANES == 2
        #define BUS_WIDTH 2
        #define NUM_LANES "2" 
[... 394 lines cut ...]
                        status = "okay";
                        reg = <1>;
                        #address-cells = <1>;
                        #size-cells = <0>;
#if VC_MIPI_MANUFACTURER == 1
                        vc_mipi@1a {
#else
                        vc_mipi@60 {
#endif
                                reset-gpios = <&tegra_main_gpio CAM1_PWDN GPIO_ACTIVE_HIGH>;
                        };
                };
#endif
        };

        gpio@2200000 {
                camera-control-output-low {
                        gpio-hog;
                        output-low;
                        gpios = <CAM0_PWDN 0 CAM1_PWDN 0>;
                        label = "cam0-pwdn","cam1-pwdn";
                };
        };
};
```

bazo80, 2025-01-06:
Hello @mstanisz0,

I've used your setup (dtsi+OrinNano+Devkit+IMX900 Rev02)
and the gstreamer-call:
```
gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! 'video/x-raw(memory:NVMM),width=2048,height=1536,framerate=30/1,format=NV12' ! nvvidconv ! fpsdisplaysink video-sink=xvimagesink sync=false -v
```
worked. But I was directly on the device, not via ssh connection.
The ./demo.sh --argus call did not work. So, we have to adjust the demo.sh script.

What is
```
nvargus_nvraw --lps
```
reporting?

And can you post the last lines of
```
journalctl -f
```
while attempting to run the gst-launch command?

best regards

mstanisz0, 2025-01-07:
Hi @bazo80 thanks for the help.

After I posted, I found out that my updated DTB wasn't being pointed to in the FDT line per the [Device Tree Flash](https://github.com/VC-MIPI-modules/vc_mipi_nvidia/blob/master/doc/DEVICE_FLASHING.md#device-tree-flash) instructions.  Updating the FDT bricked my setup, so I re-flashed with my Nano starting with my device tree modified for gstreamer and it booted.

I also switched to running on the Nano directly rather than SSH.

I tried your gstreamer command and it worked perfectly, thank you!  

Here's the information you requested:

```
vc@nvidia:~$ nvargus_nvraw --lps
nvargus_nvraw version 1.14.0
Number of supported sensor entries 1
Entry  Source Mode      Uniquename             Resolution   FR  BitDepth  Mode
Index  Index  Index                                             CSI Dyn   Type
  0      0      0          jakku_rear_vc_mipi   2048x1536  112  10  10    Bayer
```

```
-- Logs begin at Mon 2024-06-17 20:29:40 UTC. --
Jan 07 17:46:30 nvidia systemd[1627]: Started Tracker metadata database store and lookup manager.
Jan 07 17:46:30 nvidia dbus-daemon[1681]: [session uid=1000 pid=1681] Activating via systemd: service name='org.freedesktop.Tracker1.Miner.Extract' unit='tracker-extract.service' requested by ':1.76' (uid=1000 pid=2575 comm="/usr/libexec/tracker-miner-fs " label="kernel")
Jan 07 17:46:30 nvidia systemd[1627]: Starting Tracker metadata extractor...
Jan 07 17:46:31 nvidia tracker-extract[3951]: Set scheduler policy to SCHED_IDLE
Jan 07 17:46:31 nvidia tracker-extract[3951]: Setting priority nice level to 19
Jan 07 17:46:31 nvidia dbus-daemon[1681]: [session uid=1000 pid=1681] Successfully activated service 'org.freedesktop.Tracker1.Miner.Extract'
Jan 07 17:46:31 nvidia systemd[1627]: Started Tracker metadata extractor.
Jan 07 17:46:41 nvidia systemd[1627]: tracker-extract.service: Succeeded.
Jan 07 17:47:01 nvidia tracker-store[3944]: OK
Jan 07 17:47:01 nvidia systemd[1627]: tracker-store.service: Succeeded.
Jan 07 17:47:11 nvidia systemd[1627]: Started VTE child process 3982 launched by gnome-terminal-server process 3633.
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: === nvargus_nvraw[3902]: CameraProvider initialized (0xffffa8ebbe90)=== nvargus_nvraw[3902]: CameraProvider destroyed (0xffffa8ebbe90)=== nvargus_nvraw[3902]: Connection closed (FFFFAE020900)=== nvargus_nvraw[3902]: Connection cleaned up (FFFFAE020900)=== gst-launch-1.0[3990]: Connection established (FFFFAE020900)OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module0
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module1
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclHwGetModuleList: No module data found
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclHwGetModuleList: No module data found
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: OFParserGetVirtualDevice: NVIDIA Camera virtual enumerator not found in proc device-tree
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: ---- imager: No override file found. ----
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: (NvCamV4l2) Error ModuleNotPresent: V4L2Device not available (in /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function findDevice(), line 256)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: (NvCamV4l2) Error ModuleNotPresent:  (propagating from /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function initialize(), line 60)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: (NvOdmDevice) Error ModuleNotPresent:  (propagating from dvs/git/dirty/git-master_linux/camera-partner/imager/src/devices/V4L2SensorViCsi.cpp, function initialize(), line 111)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclDriverInitializeData: Unable to initialize driver v4l2_sensor
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclInitializeDrivers: error: Failed to init camera sub module v4l2_sensor
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclStartPlatformDrivers: Failed to start module drivers
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclDriver_V4L2_Focuser_Stub_Close: Invalid NULL input pPclDriver
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclStateControllerOpen: Failed ImagerGUID 1. (error 0xA000E)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: NvPclOpen: PCL Open Failed. Error: 0xf
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: SCF: Error BadParameter: Sensor could not be opened. (in src/services/capture/CaptureServiceDeviceSensor.cpp, function getSourceFromGuid(), line 726)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: SCF: Error BadParameter:  (propagating from src/services/capture/CaptureService.cpp, function addSourceByGuid(), line 453)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function addSourceByIndex(), line 347)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function getSource(), line 519)
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: ---- imager: No override file found. ----
Jan 07 17:47:18 nvidia nvargus-daemon[1025]: LSC: LSC surface is not based on full res!
[... cut ...]
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 118
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 119
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 120
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 121
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 122
Jan 07 17:47:22 nvidia kernel: vc_mipi 10-001a: vc_sen_set_gain(): Set sensor gain: 129
```

bazo80, 2025-01-10:
Hello @mstanisz0 ,

I guess, we can close the issue.

mstanisz0, 2025-01-23:
Hi @bazo80 ,

Thanks again for your help.  Streaming is working great!  I'm now trying to generate a single frame and convert the RAW image.  I'd use gstreamer, but I need to be able to adjust the exposure time and I cannot seem to get that to work.  I did it with `v4l2-ctl` and `ffmpeg` using these commands:

```
 v4l2-ctl -d /dev/video0 --set-fmt-video=width=2048,height=1536,pixelformat=RG10 --set-ctrl=exposure=10000,bypass_mode=0 --stream-mmap --stream-count 1 --stream-to=image.raw
ffmpeg -y -s:v 2048x1536 -pix_fmt bayer_rggb16le -i image.raw image.png
```

However, the image looks a bit off from the gstreamer generated one.  I take it that's due to using 16-bit instead of 10-bit bayer conversion.  Do you have a command(s) that would generate a single frame from the IMX900 and convert the RAW image to a PNG?

Matt