VC-MIPI-modules/vc_mipi_nvidia #131: Can not see grey pixel formate on R35.4.1

Ray9090, 2025-01-24:
I am using R354.1 on Jetson Orin Nano Devkit using Yocto build tools and IMX 252 camera. I found out there is only one-pixel format. First, I tried to set it from the terminal 'v4l2-ctl -d /dev/video0 --set-fmt-video=pixelformat=GREY'.  Later I see in tegra234-camera-vc-mipi-cam.dtsi, the pixel_phase  = "rggb";

I have set it to 'grey' and flashed the image, but it stuck on the flashing process. 

Should not there be a formate list? (like mention as Image Streaming in GREY, Y10, Y12, SRGGB8, SRGGB10, SRGGB12, SGBRG8, SGBRG10, SGBRG12 format). If not, then did I do right?

Ray9090, 2025-01-28:
Sometimes, it flashed but the vc driver can not initialized and got this 

root@jetson-orin-nano-devkit-nvme:~# dmesg | grep vc_mipi
[    5.031067] vc_mipi 9-001a: vc_probe(): Probing UNIVERSAL VC MIPI Driver (v0.18.0)
[    5.406598] vc_mipi 9-001a: vc_init_io(): Init trigger and flash mode
[    5.420527] vc_mipi 9-001a: vc_init_frmfmt(): Init frame (width: 2048, height: 1536, fps: 0)
[    5.420534] vc_mipi 9-001a: vc_init_binning(): Init binning modes
[    5.450963] vc_mipi 9-001a: Unsupported pixel format
[    5.455950] vc_mipi 9-001a: Failed to read mode0 image props
[    5.461727] vc_mipi 9-001a: Could not initialize sensor properties.
[    5.468026] vc_mipi 9-001a: Failed to initialize vc_mipi
[    5.473538] vc_mipi 9-001a: vc_probe(): Tegra camera device registration failed
[    5.486434] vc_mipi 10-001a: vc_probe(): Probing UNIVERSAL VC MIPI Driver (v0.18.0)
[    5.877066] vc_mipi 10-001a: vc_init_io(): Init trigger and flash mode
[    5.897016] vc_mipi 10-001a: vc_init_frmfmt(): Init frame (width: 2048, height: 1536, fps: 0)
[    5.905329] vc_mipi 10-001a: vc_init_binning(): Init binning modes
[    5.917056] vc_mipi 10-001a: Unsupported pixel format
[    5.922303] vc_mipi 10-001a: Failed to read mode0 image props
[    5.927905] vc_mipi 10-001a: Could not initialize sensor properties.
[    5.934292] vc_mipi 10-001a: Failed to initialize vc_mipi
[    5.939631] vc_mipi 10-001a: vc_probe(): Tegra camera device registration failed

bazo80, 2025-02-19:
Hello @Ray9090 , 

sorry for the late response.
The v4l2-ctl call
```
v4l2-ctl --list-formats-ext
```
will only show the formats which are defined in the device-tree modeX.

For the IMX252 you could extend your dtsi settings by two additional modes

<pre>
                                mode0 {

...
//                                        mode_type                = "bayer";
//                                        pixel_phase              = "rggb";
                                        csi_pixel_bit_depth      = "10";
                                        pixel_t                  = "y10";
...
                                };

                                mode1 {
...
//                                        mode_type                = "bayer";
//                                        pixel_phase              = "rggb";
                                        csi_pixel_bit_depth      = "12";
                                        pixel_t                  = "y12";
...
                                };

                                mode2 {
...
//                                        mode_type                = "bayer";
//                                        pixel_phase              = "rggb";
                                        csi_pixel_bit_depth      = "8";
                                        pixel_t                  = "gray";
...
                                };

</pre>

Although the pixel_t is marked as deprecated in the NVIDIAs developer guide, it is still in use. The csi_pixel_bit_depth is an obligation, but the mode_type and the pixel_phase can be left out when the pixel_t is given.

My IMX568 output:
<pre>
vc@nvidia:~$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete <del>2464x2064</del>
			Interval: Discrete infs (0.000 fps)
	[1]: 'Y10 ' (10-bit Greyscale)
		Size: Discrete <del>2464x2064</del>
			Interval: Discrete infs (0.000 fps)
	[2]: 'Y12 ' (12-bit Greyscale)
		Size: Discrete <del>2464x2064</del>
			Interval: Discrete infs (0.000 fps)
</pre>
  > Be aware, the --list-formats-ext call doesn't reflect the modeX order.

After that dtsi changes you can call:

<pre>
v4l2-ctl -d /dev/video0 --set-fmt-video=width=2048,height=1536,pixelformat="GREY" --stream-mmap
v4l2-ctl -d /dev/video0 --set-fmt-video=width=2048,height=1536,pixelformat="Y10 " --stream-mmap
v4l2-ctl -d /dev/video0 --set-fmt-video=width=2048,height=1536,pixelformat="Y12 " --stream-mmap
</pre>

best regards

noob502, 2025-04-23:
@bazo80 Thanks, it works after modification accoring to your suggestion. The issue can be closed now.