VC-MIPI-modules/vc_mipi_nvidia #132: VC IMX568 DeviceTree

nirushka, 2025-01-29:
Hi,

I am using IMX568 sensor On a custom Nvidia NX Xavier carrier board with Jetpack 5.0.2.
I would like to know which values should i put in the device tree.

I have an issue that my video seems to be wrong. I can't see a clear image.

VC_MIPI_METADATA_H = 4 // Tried 2 also
NUM_LANES = 2

My current device tree params are:

                physical_w              = "6.752";
                physical_h              = "5.655";

                    num_lanes                = NUM_LANES;
                    tegra_sinterface         = "serial_a";
                    embedded_metadata_height = VC_MIPI_METADATA_H;
                    readout_orientation      = "0";

                    // ----------------------------------------------------
                    // If you want to use GStreamer with nvarguscamerasrc
                    // you have to adjust this settings. 
                    active_w                 = "2464";
                    active_h                 = "2064";
		    active_l 		     = "0";
		    active_t 		     = "0";

                    mode_type                = "bayer";
                    pixel_phase              = "rggb";
		    csi_pixel_bit_depth      = "10";
                    min_gain_val             = "0";         // mdB
                    max_gain_val             = "48000";     // mdB
                    step_gain_val            = "100";       // mdB
                    default_gain             = "0";         // mdB

                    min_exp_time             = "1";         // us
                    max_exp_time             = "10000";   // us
                    step_exp_time            = "1";         // us
                    default_exp_time         = "1000";     // us

                    min_framerate            = "0";         // mHz
                    max_framerate            = "41300";     // mHz
                    step_framerate           = "100";       // mHz
                    default_framerate        = "41300";     // mHz
                    // ----------------------------------------------------

                    gain_factor              = "1000";
                    exposure_factor          = "1000000";
                    framerate_factor         = "1000";
                    inherent_gain            = "1";
                    min_hdr_ratio            = "1";
                    max_hdr_ratio            = "1";

                    line_length              = "0";
                    phy_mode                 = "DPHY";
                    discontinuous_clk        = "no";
                    mclk_khz                 = "24000";
                    pix_clk_hz               = "74250000";
                    mclk_multiplier          = "9.33";
                    cil_settletime           = "0";
                    dpcm_enable              = "false";

nirushka, 2025-01-30:
The problem I am seeing:

https://github.com/user-attachments/assets/53d07bc9-c1f6-4411-b1e3-644c7fb2e14c

bazo80, 2025-02-12:
Hello @nirushka ,

the v4l2-ctl is recommending the metadataheight value. The PregiusS (IMX568) is needing a metadataheight value of 1 in order to stream correctly, although the datasheet is stating otherwise (#103). We have adjusted this value in the dtsi files for the next revision of the driver. Other streaming tools are a little bit less restrictive.

We had the same kind of tear-down-effect for the PregiusS on nvidia platforms, so that we had to set the end of active video bit. On which driver version are you running? 
Maybe you took an older version, where this bit is not set.

Btw: which streaming tool have you used?

best regards

bazo80, 2025-02-12:
From the video you posted, it looks a little bit pale.
There is the black_level control the ISP (gstreamer) is not aware of.
This control is set during initialization by the driver and the v4l2-ctl -l is showing 0 as a value.
So, you could set
<pre>
v4l2-ctl -c black_level=1
v4l2-ctl -c black_level=0
</pre>

The black_level gives a little offset to the pixels, the ISP thinks it is too bright and tries to regulate down the exposure/gain, but the offset stays.
So resetting the black_level would give you a more saturated image.

The other way would be to place a so called camera_override.isp file into the /var/nvidia/nvcam/settings/ directory.
In the override file, there is the possibility to handle/correct certain values like colors or autowhite balancing.
I have to ask the colleagues, whether we have something here.

best regards

nirushka, 2025-02-13:
Hi Bazo, 

I changed the black_level and indeed, it seems way less pale.
Do you have an example to such camera_override.isp file for VC sensors?

bazo80, 2025-02-18:
Hello @nirushka ,

regarding the camera_override.isp file you can contact the official VisionComponents support.
support@vision-components.suocrm.eu
You can reference this github issue into that support request.

best regards

VC-Support, 2025-02-26:
Tuning file sent per email.

closed 2025-02-26 by VC-Support