VC-MIPI-modules/vc_mipi_nvidia #133: 4 Lans CSI0 with IMX252 is not working

noob502, 2025-01-30:
Hello all,

We have developed a customer carrier board that is very similar to the Jetson Orin Nano devkit, with a custom image created using the Yocto tool. We are using IMX252 camera sensor and Vision Component driver.

We have followed the Jetson Nano product design guide for 4 lanes cames.

[image]

We can see both cameras at /dev/video* and get the basic info on each camera, but we cannot capture any frame with CSI0.

According to that, we have modified the device tree following:

`
/*
 * Copyright (c) 2024, Vision Components GmbH.  All rights reserved.
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful, but WITHOUT
 * ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for
 * more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#include <dt-bindings/media/camera.h>
#include <dt-bindings/pinctrl/pinctrl-tegra.h>

// ------------------------------------------------------------------------------------------------
//  Driver Configuration for NVIDIA Jetson Orin Nano 
// ------------------------------------------------------------------------------------------------
#define VC_MIPI_CAM_0       1   // 1: Enabled, 0: Disabled  (serial_a => CSI Port 0 => VI Stream 0)
#define VC_MIPI_CAM_1       1   // 1: Enabled, 0: Disabled  (serial_c => CSI Port 2 => VI Stream 2)
// ------------------------------------------------------------------------------------------------
//  Supported number of lanes
// -----+------------------------------------------------------------------------------------------
//  1   | OV7251, IMX296, IMX297
//  2   | OV9281, IMX264, IMX265
//  2,4 | IMX178, IMX183, IMX226, IMX250, IMX252, IMX273, IMX290, IMX327, IMX335, IMX392, 
//      | IMX412, IMX415, IMX462, IMX565, IMX566, IMX567, IMX568, IMX900
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_LANES       4   // 1, 2, 4 Lanes
// ------------------------------------------------------------------------------------------------
//  Embedded Metadata Height
// -----+------------------------------------------------------------------------------------------
//  0   | IMX178, IMX183, IMX226, IMX250, IMX252, IMX264, IMX265, IMX273, IMX392, OV7251, OV9281
//  1   | IMX290, IMX327, IMX335, IMX415, IMX462, IMX900
//  2   | IMX296, IMX297, IMX412
//  4   | IMX565, IMX566, IMX567, IMX568
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_METADATA_H  "0" // "0", "1", "2", "4" Lines of meta data
// ------------------------------------------------------------------------------------------------
//   Sensor Manufacturer
// -----+------------------------------------------------------------------------------------------
//   1  | Sony Sensor (IMX)
//   2  | Omni Vision Sensor (OV)
// -----+------------------------------------------------------------------------------------------
#define VC_MIPI_MANUFACTURER 1
// ------------------------------------------------------------------------------------------------
//  GStreamer Support
// ------------------------------------------------------------------------------------------------
//  If you want to use GStreamer with nvarguscamerasrc you have to adjust this settings in the 
//  device tree below. The sections which have to be modified are marked by a comment. 
//  To find the correct parameter values please follow the instruction in the main README.md of 
//  this repository https://github.com/VC-MIPI-modules/vc_mipi_nvidia#gstreamer-support
[... 407 lines cut ...]
                        status = "okay";
                        reg = <1>;
                        #address-cells = <1>;
                        #size-cells = <0>;
#if VC_MIPI_MANUFACTURER == 1
                        vc_mipi@1a {
#else
                        vc_mipi@60 {
#endif
                                reset-gpios = <&tegra_main_gpio CAM1_PWDN GPIO_ACTIVE_HIGH>;
                        };
                };
#endif
	};

        gpio@2200000 {
                camera-control-output-low {
                        gpio-hog;
                        output-low;
                        gpios = <CAM0_PWDN 0 CAM1_PWDN 0>;
                        label = "cam0-pwdn","cam1-pwdn";
                };
        };
};
`

noob502, 2025-02-04:
@VC-Support please can you tell me, is the modification is correct or not ?

noob502, 2025-02-04:
I have tried ./quickstart.sh method with necessary changes for example 4 lanes, for CAM0: port-index = <0x00>,  tegra_sinterface = "serial_a" and for CAM1: port-index = <0x02>,  tegra_sinterface = "serial_c"

v4l-ctls was missing so I manually installed it yet, got this error 

jetson@nvidia:~$ v4l2-ctl -d /dev/video1 --stream-mmap --stream-count=100
[   65.760426] WARNING: CPU: 0 PID: 2059 at /home/master/vc_mipi/vc_mipi_nvidia/build/Xavier_35.4.1/Linux_for_Tegra/source/public0
[   65.782553] ---[ end trace fc8412552347a3c0 ]---
[   65.787505] tegra194-vi5 13e40000.host1x:vi1@14c00000: failed to register control callback
[   65.796029] tegra-camrtc-capture-vi tegra-capture-vi: vi capture setup failed
[   65.803532] WARNING: CPU: 0 PID: 2059 at /home/master/vc_mipi/vc_mipi_nvidia/build/Xavier_35.4.1/Linux_for_Tegra/source/public8
[   65.824643] ---[ end trace fc8412552347a3c1 ]---
[   65.829432] tegra194-vi5 13e40000.host1x:vi1@14c00000: IVC control submit failed
[   65.837044] tegra194-vi5 13e40000.host1x:vi1@14c00000: csi_stream_release: failed to disable nvcsi tpg on stream 2 virtual cha0
                VIDIOC_STREAMON returned -1 (No such device)

bazo80, 2025-02-12:
Hello @noob502 ,

my first thought was the lane_polarity. So I tried to reproduce the problem on JNX42 with your dtsi file, but it was working. When I changed the lane_polarity, I just got a timeout, but not the same error message as you got. Then, I've seen your thread in the nvidia forum. That means, you took the nvme from your working JNX42 into your custom board and it succeeded?

best regards

bazo80, 2025-02-12:
@noob502 ,

btw: the pixelclock value of 600000000 for your 10 Bit setup is ok. Since the IMX252 is running with 1500 Mbps 
4*1500000000/10=600000000