VC-MIPI-modules/vc_mipi_nvidia #160: Support for reCompute J30 with Jetson Nano Orin 8GB

Mahr-lauber, 2025-04-24:
Hi,

I would like to use your cameras on the aformentioned development kit.
I already tried the quick start script in the hopes the nvidia development kit (I think it is 14 in your script) is similar enough to this board to work but no dice.

Would it be possible to get support fo this board and if so, on what time scale?

pmliquify, 2025-04-28:
Hello @Mahr-lauber ,

Please contact VC (www.vision-components.com/en/contact/) to discuss whether and how we can support you. The aforementioned dev kit is not currently planned.

Regards,
Peter

Mahr-lauber, 2025-04-28:
Hi Peter,

contact request was sent, sould this be closed or shout it remain open until support for that board has been established?

closed 2025-04-28 by pmliquify